libretro/Lakka #144: Provide prebuilt image for AppleTV (or create a howto)

mojo2012, 2015-01-04:
Hi guys,

first, great what you're trying to achieve here. This is awesome. But it would be much more awesome, if I could use it on my old 1st gen AppleTV.

I'm trying to boot it since a few days - so far without any luck. I tried to use the atv-bootloader (https://code.google.com/p/atv-bootloader/) and the prebuilt image from openelec.
Openelec boots fine, but is not what I want.

So far openelec works like this:
* ATV EFI boots MACH_KERNEL, a linux kernel embedded in a fake mach_kernel, maybe combined with an initramfs.
* The openelec root partition is in a SYSTEM image. The kernel (or EFI?) mounts the SYSTEM image and the kernel boots it.

Basically what I can see is that lakka boots in a quite similar way. We have a KERNEL and a SYSTEM image. I suppose KERNEL also includes initramfs? But what is missing, is the fake mach_kernel.

Maybe there is a way for you guys to see how openelec builds the image? I have Mac here but I can try to build stuff in a VM, if necessary. Any help is appreciated. Thanks!

mojo2012, 2015-01-04:
I think I got a step further. I'm using the openelec boot stick, but I'm booting into lakka's SYSTEM image. Though I can see the openelc boot image on screen, the messages seem to be from the lakka SYSTEM image.
The kernel output stops at "Reached target retroarch". There is no GUI, it's not pingable and no SSH.

Is there a way to debug this?

mojo2012, 2015-01-04:
I think the problem is the MACH_KERNEL from openelec, which has no KMS/DRM or it's screwed up somehow.
I get the following message when I try to start retroarch from command line:
Couldn't find a suitable DRM device.

I'm gonna try to create my own MACH_KERNEL (linux mach kernel hybrid) out of the KERNEL included in the lakka distro.

mojo2012, 2015-01-04:
I managed to copy over the nvidia drivers, and it seems that they are loaded. But I think even the binary nvidia drivers don't have KMS/DRM support.
Maybe tomorrow I'll try to compile lakka byself without KMS support and try to use X11, which should work as it works on the openelec distro.

Kivutar, 2015-01-05:
Hello, I think you could try to use nouveau, it works not bad on nvidia cards and has KMS support.
However, nouveau should work out of the box if you're using the generic SYSTEM.
I saw two other bug reports about nouveau not being loaded properly, so your problem is maybe just this bug. https://github.com/lakkatv/Lakka/issues/137

mojo2012, 2015-01-05:
I'll try the nouvea driver, but up till now I can't use the included kernel. Where do i get a compiled nouveau driver module?

Kivutar, 2015-01-06:
Nouveau was broken on our recent builds, see bug https://github.com/lakkatv/Lakka/issues/141
I uploaded a new nightly with nouveau fixed (for 64bit only for now, the 32bit one is still building...)

mojo2012, 2015-01-06:
cool, I'll try it asap. Is the module built separately or compiled in the kernel? I'm kinda fixed on kernel 3.16 so I guess the modules are not loaded, cause they are built for 3.17.x. Is there a way to incorporate the mach_kernel building process from openelec too?

Kivutar, 2015-01-06:
Do you have any OpenELEC documentation that explains why this mach_kernel is needed?

mojo2012, 2015-01-06:
no documentation, but afaik the EFI can only boot appletv's own boot.efi file, which only can load a mach kernel. So the guys over at https://code.google.com/p/atv-bootloader/ created a way to put a linux kernel into a mach kernel. openelec uses this way to create a appletv image. I guess the make scripts could be used for lakka too

Kivutar, 2015-01-06:
I wonder why, some month ago, there was an ATV project in OpenELEC, then they removed it. I thought it was just merged in the Generic project.
How do they build recent image for ATV, where is this code?

mojo2012, 2015-01-07:
I think I've installed openelec 4.x. I can find the makescripts for you!

Kivutar, 2015-01-07:
Yes, in the branch 4.2 of their repo, you can grep for ATV in config, scripts and package, and you will know what files to read to understand how they proceed to automate the build for ATV.

mojo2012, 2015-01-07:
Im currently trying to build openelec. but somethings going wrong here. The make script can't download http://ftp.gnu.org/gnu/make/make-4.0.tar.bz2. So I've to make a break here. Maybe it'll work in a few hours.
I think it's not to hard to make a mach_kernel additionally to your normal build process. It would be nice to have it.

Kivutar, 2015-01-09:
You're building the old branch? If so, bugs like this can appear.

I suggest you to just grep ATV in this old branch to understand how it works, and report the changes to the branch master of Lakka. This way you wont get thos anoying bugs.

mojo2012, 2015-04-14:
I already tried to grep ATV but didn't really get any useful results. I didn't have any time for this stuff in the last months.
Maybe I can investigate this in the next weeks. All in all I think an old AppleTV would be the perfect lakka device.

Grintss, 2016-03-05:
I know it's been almost a year, but do you still have any interest? I'd love to put this on my old ATV!

Kivutar, 2016-03-06:
Hi @Grintss 

If somebody is OK to send me an old apple TV, I can take a look at this port.

Doing *blind* ports for all hardware is counter productive, since our todo list is quite furnished: Odroid-C2, RPi3, Pine64, OrangePiPC...

mojo2012, 2016-03-25:
Hi,

as soons as I am home I can send you my AppleTV.
It is jailbroken, soit should be fairly easy to experiment with it.
Can you contact me for address derails?

gouchi, 2016-03-25:
Please bear in mind that AppleTV 1st generation is using [Nvidia GeForce Go 7300](https://en.wikipedia.org/wiki/Apple_TV#Technical_specifications) ([NV40](https://nouveau.freedesktop.org/wiki/CodeNames/#NV40) family) and according to the [feature matrix](https://nouveau.freedesktop.org/wiki/FeatureMatrix), audio hdmi is not supported and tv out support is mostly done.

mojo2012, 2016-03-25:
hdmi sound out works under appletv osx. So the chip is at least supporting it.
Wouldnt it be possible to use component or optical sound ouput?

gouchi, 2016-03-25:
The other components should work fine.

Kivutar, 2016-06-10:
What you can do is taking the LibreELEC branch that supports AppleTV, and add the lakka and libretro folders in the package directory. I tested it with MXQ and OdroidC2 and it worked, as a temporary solution.

benjamin-army, 2016-08-09:
I don't think LibreELEC branch supports Apple TV 1st as the main dev doesn't make it top of his to-do list

https://forum.libreelec.tv/thread-3-page-2.html

Kivutar, 2016-10-02:
They have an unnoficial branch for Apple TV https://github.com/chewitt/OpenELEC.tv/tree/appletv-7.0

Kivutar, 2016-11-15:
I close this for inactivity

closed 2016-11-15 by Kivutar