libretro/Lakka #150: Lakka menu is slow on RPi

Kivutar, 2015-01-11:


ProfessorKaos64, 2015-02-06:
I can confirm this, just did an install tonight on a Model B

Kivutar, 2015-02-20:
This bug also affects RPi2

blockfeed, 2015-02-21:
@ProfessorKaos64 

To disable the background wallpaper, SSH into the device and change the following in /storage/.config/retroarch/retroarch.cfg:

menu_wallpaper = "/dev/null"

You will likely also need to disable "Configuration save on exit" under General Options as well, then restart Retroarch. This setting will revert back to enabled on next load and maintain the blank wallpaper setting.

Kivutar, 2015-02-22:
This will not improve performance. The image will not be drawn, but the performance issue is in the drawing of the 3d quad, not the texture itself.

Kivutar, 2015-03-26:
The new delta timing feature in the menu makes this bug less noticeable, but it's still there.

Kivutar, 2015-04-07:
The menu is less slow since the block text rendering optimization has been added to RetroArch. Almost unnoticeable now.

ProfessorKaos64, 2015-04-07:
Nice, i'll have to test this later. Thank you

Kivutar, 2015-06-03:
Fixed.

closed 2015-06-03 by Kivutar

gabsprates, 2016-06-12:
How I can disable the 3d quad?

Kivutar, 2016-06-12:
Go in Settings > Menu > Menu Shader Pipeline, set it to OFF

Kivutar, 2016-06-12:
Also disable the text shadows, if any

gabsprates, 2016-06-12:
THANK YOU! haha