libretro/Lakka #152: Unable to connect bluetooth controller

Atarian2600, 2015-01-11:
I have an NES30 bluetooth controller that I love, but I can't get it to connect in lakka. I've been pulling my hair out over this.

If i attempt to run bluetoothctl, it doesn't accept any input. I wanted to see if I could find anything out with hcitool, but it wasn't listing any devices. I was able to turn on my bluetooth card with hciconfig (hciconfig hci0 up), and then i can scan with hcitool. It's able to see my controller, but I have no idea how to connect.

Even after turning the bluetooth card on, bluetoothctl doesn't accept any input.

Any ideas?

Also, after rebooting, I have to redo the hciconfig thing to turn the bluetooth card back on. Is there any way to keep it enabled by default?

I'm on an intel NUC using the latest x86_64 nightly.

Thanks!

Kivutar, 2015-01-12:
http://www.danielhnyk.cz/blog/view/bluetooth-linux-indecipherable-mystery
Here I read that this inability to type in bluetoothctl is because the service is not started.
I will try to fix that on my side.

duduke, 2015-01-14:
It seems that bluetoothd is not set as a service and is therefore not started by default.
You can start it with "/usr/lib/bluetooth/bluetoothd &" and then bluetoothctl will work.
I think this is an OpenElec issues as I see people complaining about it in the OpenElec forum as far back as 4.1.4.

duduke, 2015-01-14:
It seems that /usr/lib/systemd/system/bluetooth.service relies on whether /storage/.cache/services/bluez.conf exists to start the service - it does not exist by default.
do:
touch /storage/.cache/services/bluez.conf
and then:
systemctl start bluetooth.service 
and then to verify that bluetoothd is started:
ps aux | grep bluetoothd
--- it works for me.

Atarian2600, 2015-01-14:
That seems to have the service running. Now I just have to figure out how to actually pair the controller.

Thanks!

duduke, 2015-01-14:
Try the following:
bluetoothctl
power on
agent on
default-agent
pairable on
discoverable on
scan on

Then turn-on the NES30 controller, and keep typing devices within the bluetoothctl tool to see new devices.
once you see the the new device and its bluetooth MAC Address do:
trust MACADDRESS
and then:
pair MACADDRESS
and that should be it.

Atarian2600, 2015-01-14:
Thanks for all your help! That definitely gets it paired. I don't think it's being seen as a controller though. I can navigate the menus with the arrows, but that's it. And when I look at the input devices in the lakka settings, it isn't showing up. When I run retroarch-joyconfig it wants to configure my wireless xbox 360 adapter, not the bluetooth controller. Maybe I have to add the binds to my config as if its a bluetooth keyboard?

duduke, 2015-01-14:
I also have the NES30 controller. What mode are you using?
It should be set to Mode1 (Joystick mode) by pressing the START button to turn it on. 
On modes 2-3 (Start+B, Start+B+R) it binds to keyboard keys which need to be mapped in retroarch input settings.

Atarian2600, 2015-01-14:
I'm using mode 1 (just the start button). I'm only able to use the d-pad.

duduke, 2015-01-14:
Try changing the input autoconfig to off and manually editing the mappings.

Atarian2600, 2015-01-15:
It still isn't listing the NES30 under the inputs section of the lakka settings menu, so i dont think i can do any mapping with it.

Kivutar, 2015-01-15:
I don't think that the menu input section is showing any joypad names anyway. I wonder why when you do a retroarch-joyconfig -a you see an Xbox name.

Another question, when you plug your joypad, you're supposed to see a yellow message in retroarch, what does it says?

duduke, 2015-01-15:
It's over bluetooth, and I don't see a yellow message (like I do on a 360 controller).
Running retroarch-joypad from the command line is not recognizing the controller.
I'm inclined to think this is more related to issues with the NES30/FC30 controller as typing 'dmesg' shows the following output for the controller:
[   60.892158] input: 8Bitdo NES30 GamePad as /devices/pci0000:00/0000:00:04.0/usb2/2-1/2-1:1.0/bluetooth/hci0/hci0:12/0005:0000:0000.0005/input/input17
[   60.893038] hid-generic 0005:0000:0000.0005: input,hidraw4: BLUETOOTH HID v0.00 Keyboard [8Bitdo NES30 GamePad] on 00:19:86:00:2b:48

Which makes me think it is recognized as a keyboard.

Kivutar, 2015-01-16:
Yes, that's it. Well if this controller is declaring himself as a keyboard... I guess there is not much we can do. Try to configure it manually as a keyboard using rgui.

duduke, 2015-01-20:
Atarian2600 - Can you try with the latest NES30 firmware (2.52) as it is supposed to be detected as a true gamepad now.

Atarian2600, 2015-01-20:
I'll give that a shot later tonight. Thanks for the tip!

jayp76, 2015-01-23:
i tried with my FC30 controllers with the new firmware 2.52 and START+R.
unfortunately retroarch-joyconfig says "Couldn't open joystick #0."
dmesg output is:
[ 1234.331527] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[ 1234.331541] Bluetooth: HIDP socket layer initialized
[ 1234.332628] hid-generic 0005:0000:0000.0003: unknown main item tag 0x0
[ 1234.332845] input: 8Bitdo FC30 II Joystick as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/bluetooth/hci0/hci0:70/0005:0000:0000.0003/input/input15
[ 1234.333166] hid-generic 0005:0000:0000.0003: input,hidraw2: BLUETOOTH HID v0.00 Keyboard [8Bitdo FC30 II Joystick] on 00:1a:7d:da:71:0d
[ 1302.249710] hid-generic 0005:0000:0000.0004: unknown main item tag 0x0
[ 1302.249909] input: 8Bitdo FC30 II Joystick as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/bluetooth/hci0/hci0:65/0005:0000:0000.0004/input/input16
[ 1302.250197] hid-generic 0005:0000:0000.0004: input,hidraw2: BLUETOOTH HID v0.00 Keyboard [8Bitdo FC30 II Joystick] on 00:1a:7d:da:71:0d

duduke, 2015-01-23:
If you'll run:
retroarch-joyconfig -d linuxraw 
You'll be able to detect it, but unfortunately I couldn't get it to detect most of the buttons.

Kivutar, 2015-02-08:
I close this report, since bluetooth works properly. You can still reopen a new bug for your not working joypad or discuss that on the forum if it's not a bug in our code.

closed 2015-02-08 by Kivutar

97-109-107, 2015-05-27:
I managed to connect the NES30 in the 'START' mode and connected it up using duduke's commands.
I can confirm that the dpad works in the menu, but I can't configure any of the buttons to be bound to actions.

Kivutar, 2015-05-28:
Hello, please try the latest nightly. The configuration for NES30 has been updated a few days ago.

97-109-107, 2015-05-28:
Hey, thanks for the response. It was the yesterday's nightly build. I did even find the relevant, above mentioned, config file in the system, under `/etc/retroarch-joypad-autoconfig`. I was using the default driver - `udev`

Kivutar, 2015-05-28:
Ah, maybe there is more than one NES30 model... strange

97-109-107, 2015-05-28:
This is the dmesg log upon connecting the pad:

``
[  133.557752] input: 8Bitdo NES30 GamePad as /devices/soc0/soc.1/2100000.aips-bus/21f0000.serial/tty/ttymxc3/hci0/hci0:11/0005:0000:0000.0001/input/input2
``

Bluetoothctl info on the device:
````
[bluetooth]# info 61:04:76:1D:71:01 
Device 61:04:76:1D:71:01
	Name: 8Bitdo NES30 GamePad
	Alias: 8Bitdo NES30 GamePad
	Class: 0x002504
	Icon: input-gaming
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
````

I'm not familiar with *pid/vid* with bluetooth devices - what should we look at?

Kivutar, 2015-05-28:
Ah, that's it. bluetooth devices have no vid:pid. In fact, you will just see the vid:pid of your bluetooth receiver, not the one of the device, which is identified by a MAC address, since it is a network object.

I suggest you to forget about autoconfig, and bind manually in the input settings.

97-109-107, 2015-05-28:
Oh. I tried it before, the procedure was like this:

- establish Bluetooth connection
- verify that I can move around in the menu with the dpad
- with autoconfig on, I tried binding any of player 1 inputs to any button of the controller - with the type set to RetroPad and later RetroPadKeyboard
- with autoconfig off, I tried binding any of player 1 inputs to any button of the controller - with the type set to RetroPad and later RetroPadKeyboard
- restart the device, reconnect  joypad and do again, both under xmb and rgui (does it matter anymore?)

The dpad always worked for moving around the menus. No buttons (nor dpad) could be bound to any action.

Kivutar, 2015-05-28:
Ah, the one that was supposed to work was this one:

> with autoconfig off, I tried binding any of player 1 inputs to any button of the controller - with the type set to RetroPad

I read somewhere that the NES30 may need a firmware update, is it your case?

97-109-107, 2015-05-28:
I checked that, saw that I'm on the second-latest and that the changelog from the latest wouldn't indicate any relavant change (pic related). I might try doing it that later today nonetheless and report back.

[image]

Autoconfig issues aside, in general, I was also looking into buying another bluetooth pad from the same comapny - if it's a HID device - what could be preventing me from setting it up? Is lakka capable of binding any device that introduces itself as a joystick/keyboard?

Kivutar, 2015-05-28:
It's better if the joypad declare himself as a joypad.

97-109-107, 2015-05-28:
Does the dmesg log I posted indicate it does introduce itself as a joystick? below again:
`` [ 133.557752] input: 8Bitdo NES30 GamePad as /devices/soc0/soc.1/2100000.aips-bus/21f0000.serial/tty/ttymxc3/hci0/hci0:11/0005:0000:0000.0001/input/input2 ``

Kivutar, 2015-05-28:
I'm not sure. Check if something like /dev/input/js0 appear when you plug the joypad. If so, it means it is a joypad.

jayp76, 2015-05-28:
It only appears as Joystick when connected via USB. 
Here a dmesg:
[  913.058951] usb 1-2: new full-speed USB device number 4 using xhci_hcd
[  913.235108] input: FC30               FC30  Joystick as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:1235:AB11.0004/input/input17
[  913.235363] hid-generic 0003:1235:AB11.0004: input,hidraw0: USB HID v1.10 Joystick [FC30               FC30  Joystick] on usb-0000:00:14.0-2/input0

Via Bluetooth it appears as Keyboard:
[  828.117175] hid-generic 0005:0000:0000.0003: unknown main item tag 0x0
[  828.117353] input: 8Bitdo FC30 GamePad as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/bluetooth/hci0/hci0:71/0005:0000:0000.0003/input/input16
[  828.117737] hid-generic 0005:0000:0000.0003: input,hidraw0: BLUETOOTH HID v0.00 **Keyboard** [8Bitdo FC30 GamePad] on 00:1a:7d:da:71:0d

jayp76, 2015-05-28:
We'd also need an autoconfig file for the FC30 from 8Bitdo (same Firmware as the NES30) with input_device = "8Bitdo FC30 GamePad" 
and input_device = "8Bitdo FC30 II"  for 2nd Player device.

Kivutar, 2015-05-29:
OK I understand.

The thing is that RetroArch supports autoconfig files only for joypads, not keyboards.

If these pads declare themselves as keyboards, the bug may be in them, not in the system. (but I'm not 100% sure)

jayp76, 2015-05-29:
I contacted the producer of these pads via their forum. They answer normally very quick. Perhaps they can add a new mode that shows the pads as joysticks. Also awaiting Retron5 support for the pads.

jayp76, 2015-05-29:
The other thought is that retroarch can be used also via keyboard. If in the autoconfig the input driver is changed to linuxraw and the button to the corresponding keyboard button, could this work?

Kivutar, 2015-05-29:
@jayp76 I don't know.

I found this http://www.retrocollect.com/Reviews/review-8bitdos-nes30-bluetooth-gamepad.html where it is explained how to tell the device to declare as a bluetooth joystick.

jayp76, 2015-05-29:
Ahm, Bluetooth Joystick mode ist when powering up the controller with the start button. This is working mostly for Android devices. The dmesg output we provided is from this mode.

jayp76, 2015-05-29:
Powering uo with Start and pairing it with Windows, it shows as HID conform Gamecontroller. Iw works with Retroarch on Windows. So i suspect that the Openelec Bluetooth device drivers are not up to date to support this controller as a HID conform device. Just sayin. Don't want to compare Windows with Linux.

docod, 2015-06-15:
If you are using the latest firmware, there is a new "PC Joystick mode".  Hold Start + R to power it up.

Here is a quote from the firmware page:

>2.Added PC Joystick mode(power up START+R and connect to PC). D-Pad works as Joystick D-Pad, not a keyboard arrow key.

jayp76, 2015-06-15:
That's what we already tried. See also my comment from 23rd of January. It doesn't work.

97-109-107, 2015-06-16:
@docod thanks for the suggestion. Unfortunately I have tried this fw as well with no success.

docod, 2015-06-16:
Sorry I got overly excited when I read that. Then I was bummed when I got home, tried it and saw it was still showing up as a keyboard. My bad, false alarm.

Would it be possible to fix this with a driver?  I have never written a driver. I write windows software. I saw some data coming from /dev/input/js1 it seems like something could be done. I have no idea where to start, though. Any thoughts on the subject or clues how to start such a project.

jayp76, 2015-09-23:
@docod 
In case you missed that information. The 8bitdo controllers are working with a workaround.
See this thread: http://libretro.com/forums/showthread.php?t=2347&page=3

maxengel, 2016-11-23:
I can get the 8bitdo controllers working, but they are not automatically reconnected when the power is cycled. Instead, I have to re-pair the controller. Even after connecting and trusting it, it still doesn't reconnect after a power cycle.

tvhung83, 2016-12-04:
Thank you @duduke , I'm able to connect Xiaomi bluetooth controller, using your commands **AND** one more: `connect <MAC>`. It also auto-reconnect every time I power on the controller.