libretro/Lakka #158: Got RetroArch instead of Lakka UI

alexandrevicenzi, 2015-01-17:
Booted once got Lakka UI.
Booted again got RetroArch UI.

RetroArch UI doesn't work.

I'm running Lakka in BPi. Version Lakka-Bananapi.arm-devel-20150116155336-r20321-g690bef0

alexandrevicenzi, 2015-01-17:
Happened once. Just copy IMG again to SD and it's working for now. Have no idea to simulate.

Kivutar, 2015-01-17:
This bug happens when you first start Lakka, then unplug the power wire. The config file of retroarch gets corrupted or something like this. Then when you boot the second time, retroarch will not be able to open its config file, and thus will not find the joypad configs.
That's a real bug and we need to fix it.

Kivutar, 2015-11-20:
This bug should not happen more. Since we added the shutdown entry, the retroarch.cfg file will always be saved properly before you can unplug the power wire. So I close this, but feel free to reopen it if this bug happens again.

closed 2015-11-20 by Kivutar