libretro/Lakka #163: No audio in mupen64plus on imx6

Kivutar, 2015-01-21:


Kivutar, 2015-03-26:
Undefining __ARM_NEON__ and setting HAVE_NEON=0 fixes the sound issue, but we lose the speed optimization.

@twinaphex what do you think about it? Did I miss a obvious fix?

twinaphex, 2015-03-26:
Can you find out with a debugger what it fails on exactly?

I guess in the meantime, we should just undefine NEON for Raspberry Pi 2/ARM targets. But we should try to isolate the cause of what makes this happen.

Kivutar, 2015-03-26:
I don't know how to debug it with my current gdb skills. The core is just silent, it does not crash.

For now I disabled NEON on RPi2/imx6, so this core is at least playable.

I remember that this audio issue appeared long time ago, maybe one year ago, so it should be bisectable, but it would take hours.

Kivutar, 2015-12-29:
Disabling the neon cc resampler fixed this. It is incompatible with armhf for now.

closed 2015-12-29 by Kivutar