libretro/Lakka #165: [Bug] Menu too fast after exiting Atari Lynx games

jayp76, 2015-01-23:
After exiting a Atari Lynx game, the menu is very fast, you can't select accurately the next game you want to play. If you exit Lakka Menu it is OK afterwards or if you start and exit a game of another system it is OK, too.
I tested it on devel-20150116154618-r20321-g690bef0.

ProfessorKaos64, 2015-02-05:
I can confirm this bug using the `Lakka-Generic.x86_64-devel-20150130130214-r20504-g57a7349` image for PC/Linux

Kivutar, 2015-03-26:
Can you try to set the maximum run speed to 1 in the general settings, and see if it affects this bug please?

jayp76, 2015-03-26:
In the meantime i moved on to devel-20150307205641-r20834-ge8d52d6.
run speed is set to 1.0
Bug is reproducible. Also on devel-20150324235229-r20987-g7e3860c.
Lakka menu driver.
Is it normal that on xmb i can't start a game when activating run command?

Kivutar, 2015-03-26:
Yes because this xmb is old, I will upload a nigtly with an up-to-date RetroArch in the week, I still need to fix one or two regression on some platforms.

jayp76, 2015-05-16:
Bug with fast scrolling menu also present in new XMB. Lakka-Generic.x86_64-devel-20150512171823-r21002-ga5310d6

Kivutar, 2016-01-14:
Happens only when threaded video is enabled.