libretro/Lakka #168: Feedback and issues

ProfessorKaos64, 2015-02-05:
Set this up on an intel x86 chip, with an nvidia card and wire x360 controller.

1. many of the menu items under the settings crossbar don't work or do anything. I know I am using the right gamepad button (B) to select and go back (A). I am aware of Z and X for Retroarch, which work as well (**update** - solved by quitting Retroarch once).

2. Save states should default to slot 1. By default if you hit save state, it fails on the included 2048 game. That will frustrate new users who do not understand how this all works. ( **Update** - I see the option in settings for save state auto index)

3. It's not obvious how to end the game from the center button press. Pressing A works, but it would be very nice to just have an "end game" under "reset'

4. Add a ROM loader somewhere to load ROMs into Lakka. This is a just a suggestion. 

5. Like emulation station, limit the on screen systems to those that only contain roms, or make this a switchable option

6. It is not obvious that a user needs to use "Quit Retroarch" to reload Lakka to pick up the new set of ROMs copied over. A button that reads "reload ROMs" would be nice.

7. On the default Lakka setup for wired x360, B is OK, A is Cancel, seems backwards

8. Why does the User 1/2/3/4 etc device type change to the last used emulator pad type? (e.g. SNES Joypad).

9. Menu wrap around vert/horz, is this supposed to return to the top of the list when I hit down on  vertical list? Doesn't seem to work.

10. Software filter just flashes the screen, no on/off switch.

11. On startup with the wire x360 pad, the list rapidly scrolls until I stop it. Maybe a calibration setting that uses jstest or something.

12. Core Management URL, if selected, seems to just freeze the system

13. I see games seem to be detected by extension alone. This causes ROMs like Breakout.bin to be picked up in the Nintendo DS section. I have Crash Bandicoot under Sega MS/GG/MD/CD :)

14. Had a weird bug, where I exited Chips Challenge on Lynx, and then my scroll speed was much faster, making selections difficult. Restarting Retroarch fixed it. This happened on other emulators as well (Wonderswan) .Starting another ROM and exiting seemed to also fix this behavior.

15. If a ROM fails to load, provide on screen failure notice before restarting Lakka

16. An about screen in the settings menu to output version info and more.

17. Reconfigure buttons per system (in the same way Retroarch can save separate config files"

Other than these, great work! Do you have a site I can donate a few bucks? I currently maintain my own automation script to kind of do similar things with EmulationStation. I really thing you are on to something here. Glad I backed this on kickstarter.

If you want, I would be happy to spend some free time helping with documentation. The amount of options is awesome. I really like what you're on to here. I love the hot swapping controllers (wireless x360). Anyway, sorry for the long post!!!

Thanks!

Kivutar, 2015-02-05:
Hello, thanks for your feedback!
Let me try to reply point by point:

 1. This is a bug in the lakka menu driver, it is going to be replaced by xmb soon.
 2. I never noticed such a bug, for me, the default savestate is 0 and works out of the box. I will do more testing to see if it affectx XMB.
 3. This has been aften requested. But let me ask, why would you need to stop a game? While you can keep the game paused while you select another game, it will automatically close the previous one.
 4. What do you mean by ROM loader?
 5. I like this idea (as long as it is configurable) @twinaphex what do you think about it?
 6. Totally agree, I hope that we can add a define to RetroArch to allow changing small things like that. Could be implemented as a patch for now.
 7. Seems backward for european users but normal for old japanese games. Anyway, this is the default mapping in RetroArch and we can't change de default mapping. Maybe we could have input remapping for the menu, but not sure about it.
 8. I don't know, maybe because you're not supposed to plug a megadrive pad in a SNES. @twinaphex could answer this better than me.
 9. Works in XMB, the next gen menu
 10. We've not packaged them yet. There is a issue about it https://github.com/libretro/Lakka/issues/132
 11. This is a real annoying bug. We need to fix it.
 12. I don't know what to do with this yet, using that feature or hiding it. It's not an easy decision given the variety of platforms we support with our build system. There is also usability issues for users that don't use internet.
 13. Will be fixed in XMB, @twinaphex is working actively on this with the new database system and CRC checks.
 14. This bug have been declared here: https://github.com/libretro/Lakka/issues/165 ans is low on the priority list.
 15. You're right. A lot of users report bugs when they just miss a firmware. It could be nice to prevent them graphically.
 16. I agree.
 17. This works in XMB.

So before you try XMB and says that this and this don't work, please note that the XMB we ship in the nightlies is outdated ;)

Thanks for your suggestion of donating. But for now we don't accept monetary donations. However, we do accept hardware donations for the project. Thus it would be a good idea to accept money as long as we ensure that it is spent only in buying hardware for the project or that it pays for the build servers.

It's nice to hear some positive and constructive feedback. Please join #lakkatv and #retroarch on IRC. Help for documentation is always needed on both sides. Of course to write some good documentation you have to know the system very well first, like using it for weeks.

ProfessorKaos64, 2015-02-05:
Many thanks for the reply, given my lengthy initial impressions and feeback @Kivutar . Responses to your reply:

3. I mainly mean, even if its to hit "back" (A by default on x360 pads), for new combers, a stop entry would alleviate folks being confused on first use. Not necessary, but helpful for new folks I think. Or, what you could do, is just have a suggestion or caption that says "back to exit".

4. Just wondering if it's possible to load ROMs via Lakka instead of going to my other Linux boxes. Maybe an option to browse the network and select a directory to copy or something.

7. Noted and acceptable of course. I looked in the menu bindings on how to reconfigure the buttons like you would in Retroarch, and maybe I did not hit the right option to reconfigure the buttons. If you can't a future option to allow this like Retroarch would be nice.

Thank you very much for the quick response. I love the direction this is going in, and allows me to stop worrying about maintaining my own automation script/project for setting up most of this. I know XMB has some issues at the moment in nightlies, as I did try that, so I will be patient. I did add #lakkatv to my freenode favorites. After I have been using the system for a while, I will present some documentation. For now. I can start my own guide on libregeek.org or use a markdown wiki that is can be cloned here on github to make it easily accessible. I just want to start and document how to do simple things at first, help folks along.

-pk

Kivutar, 2015-02-05:
1. Sometimes you want to go to the shader options in the main menu while being able to come back to your game just by pressing the center button. If 'back' quits the game, this feature is lost.

2. Like a ROM online store? This would put us in a very uncomfortable position. Could be done for homebrews and free games however.

Thanks for the documentation effort!

twinaphex, 2015-02-05:
I think rather than an ROM online store he means something like an NAS network share or whatever.

Anyway, it's at this point where it gets a little scary trying to bake all of that into RetroArch since we do have bloat to worry about - the binary size is already at least twice as large (6MB) as it was last year (3MB). I really need to walk a tight line here between the kind of memory footprint that a PSP and an Xbox 1 have vs. say a PC or smartphone with gigabytes of RAM.

Anyway, I'm sure something will figure itself out there, but it might have to be done outside of RA.

Kivutar, 2015-02-05:
We support NFS mounting, and some of our users that own a NAS have been using that feature iirc.

On another side, we had some month ago a shell script that acted like a package manager to download contents from an http server. Just using wget and grep. It was so simple that everybody could write it so I never shiped it in Lakka.

Kivutar, 2015-02-08:
I close this, I declared some of the bugs reported there. Please try to declare only atomic bugs. Lists like that have their place on the forum, not in the tracker.

closed 2015-02-08 by Kivutar