libretro/Lakka #170: Enhancement Request : ROM filtering by letter, sorting options

ProfessorKaos64, 2015-02-05:
I know this is Lakka on top of Retroarch, but just a suggestion.

twinaphex, 2015-02-05:
Reasonable request yes. I will put it somewhere on my TODO list for the menu system.

ProfessorKaos64, 2015-02-05:
Thank you @twinaphex , keep up the great work.

Kivutar, 2015-02-05:
Another way to implement this would be to keep the list flat, but to go directly to the next/previous letter by pressing L1/R1.

arananet, 2015-02-11:
Maybe a workaround is the posiblity to create a directory tree, for example:

ROMS > NES
                SEGA
                PSX
                N64
                MAME
                ETC.

Then store the directory location into environment variables, you can filter the content to display on each emulator even using the extension as a plus ;)

Kivutar, 2015-11-20:
The L1/R1 buttons now allow you to switch to the next letter, so I close this. I agree that this is not very visual and intuitive, but it was the easiest way to implement it in the code.

closed 2015-11-20 by Kivutar