libretro/Lakka #173: Cave Story crashes on Level "Sand Zone" when Boss Fight starts

jayp76, 2015-02-06:
Cave Story crashes on Level "Sand Zone" when Boss Fight starts

[  277.414126] retroarch[573]: segfault at 38e8900 ip 00007fb41ad17176 sp 00007fff06d72ae8 error 4 in nxengine_libretro.so[7fb41acb1000+aa000]

jayp76, 2015-02-06:
Forgot to mention, tested on x86_64.

jayp76, 2015-02-06:
Crashes on Raspi port, too.

Kivutar, 2015-02-06:
Can you upload a savestate somewhere please?

jayp76, 2015-02-06:
I think its the profile.dat. Here you are. Thanks for looking into the issues Kivutar.

https://www.dropbox.com/s/v71vd5a0603kjp7/profile.dat?dl=0

Kivutar, 2015-02-06:
Thanks, tried it on my laptop with the same nxengine version and went to the boss fight without a segfault. Let me try this on the liveUSB now.

Kivutar, 2015-02-06:
OK, I was able to reproduce and fix. The version of nxengine packaged in Lakka was old. I'm uploading a new nightly with the fix. Thanks for the bug report.

closed 2015-02-06 by Kivutar