libretro/Lakka #174: BeetlePSX crashes when loading game images

jayp76, 2015-02-06:
BeetlePSX crashes when loading different game images.
Tried iso, ccd and cue formats.

[   11.675091] retroarch[471]: segfault at 0 ip 00007fc22de3ed09 sp 00007fff88f1edf0 error 4 in beetle_psx_libretro.so[7fc22dddc000+9f000]

It's odd that only "klaymen" (japanese Neververhood) is working.

Kivutar, 2015-02-06:
I got almost all my .cue loading in beetle-psx :o 
See https://www.youtube.com/watch?v=ZFWbEIA7w8Q
What hardware are you using?

jayp76, 2015-02-06:
Forgot to mention, tested on x86_64.

jayp76, 2015-02-06:
MSI Probox23 with 4GB Ram, and Intel i3 CPU.

I put the PSX Bios files under "system". Is this correct?

Kivutar, 2015-02-06:
It's the correct path yes.

Check that you have all the BIOSes https://github.com/libretro/libretro-super/blob/master/dist/info/beetle_psx_libretro.info and that their names and md5 sums match.

Check that your .cue contains the right name for the corresponding .bin.

jayp76, 2015-02-06:
Bios names and md5 sums are matching.
Also checked the cue files. Nothing wrong with the names.

jayp76, 2015-02-06:
Seem to be the images.
Beetle PSX should have ccd support but it does not work.
When i convert the ccd images to bin/cue it works mostly.

Kivutar, 2015-02-06:
Yes, mednafen PSX supports mostly bin+cue files. I was able to launch almost all my bin+cue files so far. I think that further request should be made upstream. I will close this.

closed 2015-02-06 by Kivutar