libretro/Lakka #177: GpSP whitescreen

dasfreak, 2015-02-08:
Some games (i.e. Pokemon) will result in a whitescreen.

Started with tty retroarch=0 and searched for games_config.txt but couldn't find one.

Similar issue:
https://github.com/petrockblog/RetroPie-Setup/issues/218

Tested on bananapi with newest nightly and working gba_bios (tested with other games).

Kivutar, 2015-02-08:
I loaded a rom through gpsp-libretro and saw this message: config_path is : /home/kivutar/Jeux/roms/gba/game_config.txt

So I guess you have to take this file https://github.com/libretro/gpsp/blob/master/game_config.txt and put it next to your rom.

dasfreak, 2015-02-08:
Yes, this works. Thanks!

Kivutar, 2015-02-08:
Let me close that then. I will tell urukai to document that on the BIOS doc page.

twinaphex, 2015-02-08:
Can you mention the exact game this still happened with?

Because having that file in your directory should no longer be necessary. If it does, then maybe it's not properly finding the entry of the game in the built-in database.

dasfreak, 2015-02-08:
Pokemon Feuerrot (D).gba
Pokemon Smaragd (D).gba

The Pokemon games are known for whitescreen errors. It is solved by setting "Flash 128K''

Final Fantasy I & II - Dawn of Souls.gba worked without the list.

twinaphex, 2015-02-08:
Well I already added Pokemon Emerald (G) to the gba-over list.

https://github.com/libretro/gpsp/blob/master/gba_over.h#L1536

So it should have worked without that game_list.txt file.

dasfreak, 2015-02-08:
Is it because of the german version?

Kivutar, 2015-02-08:
The (G) is for german. Can you copy the whole log of RetroArch when launching this rom please? It could show if the core goes through the end of the list without matching the ROM.

Cloudbarrage, 2015-04-19:
I can't start Pokemon Ruby or Emerald with the standard configurations, but when I change the Config Directory to a path containing the game_config.txt file I can run them.
Would be interesting to know where the default path is pointing to...

Kivutar, 2015-04-19:
I guess it's pointing to ~/.config/retroarch/

closed 2015-06-25 by Kivutar