libretro/Lakka #178: Unable to access command prompt in PC

vicmarto, 2015-02-17:
Hello.

I'm following the instructions to get a command prompt in a PC: http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface

When typing "tty retroarch=0" I get an error: "No such file or directory"

Please, look:
[image]

I need to get a command prompt to setup my wifi dongle, and then get SSH access. No ethernet in this machine, sorry.

Thanks!

Kivutar, 2015-02-24:
Look like you removed the whole line. You should not. Just add "tty retroarch=0" to the existing line without removing anything.

Kivutar, 2015-03-08:
Somebody accessed the command prompt not so long ago, so I guess it was just that.

closed 2015-03-08 by Kivutar