libretro/Lakka #180: Segfault when launching any ROM on an ATI card

Kivutar, 2015-02-20:


Kivutar, 2015-02-25:
No way to get a backtrace...

    Program received signal SIGSEGV, Segmentation fault.
    0x000000000048f218 in ?? ()
    (gdb) bt
    #0  0x000000000048f218 in ?? ()
    #1  0x000000000048f4d3 in ?? ()
    #2  0x0000000000413506 in ?? ()
    #3  0x00000000004137f4 in ?? ()
    #4  0x00000000004085ff in ?? ()
    #5  0x00007f1fdba55770 in __libc_start_main () from /lib/libc.so.6
    #6  0x0000000000406809 in ?? ()

GPDP2, 2015-02-26:
I was just going to file this as well. Lakka works well on my laptop with Intel graphics, but always fails to start a game on my desktop with AMD graphics.

aboutafter, 2015-03-01:
I can confirm it too. Both with a laptop and a desktop PC with AMD cards.

ShootHere, 2015-04-02:
Confirmed on Desktop with ATI Radeon HD 5450

rasmusfjord, 2015-04-08:
?+so any update on this

Kivutar, 2015-04-08:
This bug is still present in our latest build. I will give it another try this week, trying to get a better backtrace.
Another way would be to git bisect, but it would take ages.

rasmusfjord, 2015-04-13:
tried 3 different release prior to 20 feb but none works some dosnt even install. Hoping you guys can get it working :D since i bought controllers a small machine to run it on :D :+1:

Kivutar, 2015-05-05:
Can one of you test this image on ATI please?
http://static.kivutar.me/Lakka-Generic.x86_64-devel-20150505193651-r21001-g922391b.img.gz
You need an UEFI capable motherboard.
You can just boot it, and launch the 2048 game, if it works, then it works.

aboutafter, 2015-05-06:
No UEFI motherboards for testing here. That's probably why I can't boot the latests images from Lakka too. I hope UEFI doesn't become a must from now on because that is going to make all the old PC's useless for this.

twinaphex, 2015-05-06:
Yeah UEFI should definitely not become a requirement at least. I totally believe in giving old hardware a second lease on life through something like Lakka.

Kivutar, 2015-05-06:
I also believe in using old hardware, since it works very well. However, OpenELEC is droping both i386 and BIOS support now. So i'm considering adding this support back, or forking, or switching to buildroot. (However, let's discuss this problem in another thread and keep this thread about ATI please)

twinaphex, 2015-05-06:
Maybe after RetroArch 1.1 is out we need to re-evaluate whether or not we should do our own minimal distro or continue with the OpenElec-based distro. I don't think it's reasonable at all to be dropping BIOS and i386 support and in fact that goes against my plans to have this distro be as backwards compatible as possible. I want it to be possible to install this on old-world Macs too with PowerPC CPUs in the long run. 

But yeah, I'll digress, we can move this to another thread.

ShootHere, 2015-05-06:
tried your build but still the same result. Not sure if I have a UEFI mobo though (any way to find out?) so take it as you will.

rasmusfjord, 2015-05-06:
Ill give a try but it will be a few days before i get time.

Kivutar, 2015-05-19:
Ok you can also try this: http://static.kivutar.me/Lakka-Generic.x86_64-devel-20150518150748-r21009-gd55ab55.img.gz

This image uses X11, and does not require UEFI. Please tell me if it fixes the ATI compatibility.

aboutafter, 2015-05-19:
I can't boot from it. I'm copying the image to the USB with dd.

rasmusfjord, 2015-05-19:
I tried this release and i could run 2048 but couldnt run a snes game.

So a step futher.

aboutafter, 2015-05-20:
It kinda works, but there are some issues, I don't know if they are related to this version alone or to Lakka in general:

- SNES (both cores) and Megadrive/Master System don't work
- the performance is worse than from any GNU/Linux distribution. No idea why this can be, but I can't use any shaders, even the most simpler ones. 
- changing the default shaders with M,N buttons breaks the game.
- the laptop volume controls don't work.

I've tried successfully these systems: N64, Sony PSX, Arcade (FBA), Nintendo GBA & GB.

EDIT: that was in my laptop. In other PC connected to a TV with HDMI I don't have sound and only the GBA core works (and the 2048 game).

Kivutar, 2015-05-20:
OK, the snes games or sega games not working are due to a another bug in RetroArch and will be fixed soon.

It's good to know that ATI compatibility can be fixed using X11. It's a good diagnostic (if not a fix).

The shader issue will be easy to fix: package CG shaders for X11, package GLSL shaders for drm/kms.

What makes me worry is the bad performance. I know that X11 add some overhead (that's why we removed it in the first place). But on intel cards, I didn't notice too much difference. On what game did you notice the bad performance?

aboutafter, 2015-05-20:
Without using shaders the performance is fine, the problem is for example simply adding scanlines or other 1pass shaders, that doesn't impact the performance when using retroarch from Arch or Xubuntu but it does with Lakka. 

Anyway I'm more concerned by the fact that in the desktop PC were I tried (AMD processor and ATI 5770 card) which has better performance than the laptop, I could only run GBA games. The other cores showed a black screen for a second and went back to xmb.

twinaphex, 2015-05-20:
Sounds like a GPU-bound issue, possibly driver-related if you get bad performance just with shaders (which are entirely performed on the GPU if it didn't need pointing out). I don't own any hardware with AMD GPUs so I can't really testify to this, but there's a lot of bad press about AMD's drivers on Linux so I wouldn't rule out the possibility of it being a driver issue.

Kivutar, 2015-05-21:
@twinaphex The reason of this bad performance is that we ship glsl drivers in Lakka for the reason that drm/kms supports only those. This x11 image is here to check that the ATI compatibility is fixed by x11. I didn't take the time to package cg shaders yet. Because if I find a way to keep it pure with drm/kms instead of x11, I will do.

@aboutafter This black screen is another bug. You can workaround it by disabling OSD messages in the font settings.

I'd like sombody with an nvidia card to compare the performances of ppsspp, between this x11 image that uses the proprietary nvidia driver, and the release that uses the nouveau driver over kms. I think that the performance on GL cores could be improved significantly on nvidia if we use this proprietary driver.

closed 2015-06-03 by Kivutar

Kivutar, 2015-10-30:
Hi, could you please test those two images on ATI?
The first one have DRM/KMS, and should segfault on ATI.
The second one have DRM/KMS + a patch that may fix the problem.

http://static.kivutar.me/Lakka-Generic.x86_64-devel-20150929235257-r21175-gba5c201.img.gz
http://static.kivutar.me/Lakka-Generic.x86_64-devel-20151028051755-r21187-g48d49cd.img.gz

Thanks.

aboutafter, 2015-11-01:
The first doesn't work no, and the first time I tried the second one it didn't work either, BUT it worked great after that!

I don't know why failed the first time trying to run a game though, but it seems that the segfault is fixed now, at least with my ATI Mobility Radeon 5000 series.

And this is unrelated to this topic, but I like a lot the option to scan the disk drives in the PC when using the live mode.

Kivutar, 2015-11-01:
Ok, thanks for this precious information. Could you please confirm that with the second image (r21187) the first game launched after a fresh install always causes a segfault? Because that would be problematic..

aboutafter, 2015-11-01:
I tried with another USB drive and downloading the file again just in case. 

The first time I try to launch a game the screen goes black for a fraction of a second and shows the menu again. Then you can press "Run" again and the game starts fine. It happens only the first time you use a core.

garnieretienne, 2015-11-02:
I think this is on purpose (but annoying), it doesn't start when selecting any emulator the first time, it just remember the choice and need to be started again. I have the same behaviour on my Generic x64 build with an nvidia card.

Kivutar, 2015-11-02:
@garnieretienne selecting the core doesn't cause the menu to go black.