libretro/Lakka #181: ATI AC97 VT8233 No Sound

delta360, 2015-02-23:
With the VT8233 chipset there is no sound returned via the speakers, Alsa returns that there is no sound card even though the following outputs say otherwise.

LSMOD
Lakka:/ # lsmod
Module                  Size  Used by
snd_intel8x0           24845  0
snd_ac97_codec         99286  1 snd_intel8x0
ac97_bus                1126  1 snd_ac97_codec
snd_pcm                60823  2 snd_ac97_codec,snd_intel8x0
p54pci                  7500  0
p54common              23507  1 p54pci
mac80211              242411  1 p54common
cfg80211              192001  2 mac80211,p54common
k8temp                  3119  0
firewire_ohci          26141  0
firewire_core          42605  1 firewire_ohci
crc_itu_t               1283  1 firewire_core

LSPCI
Lakka:/ # lspci | grep -A 20 Audio
00:11.5 Multimedia audio controller: VIA Technologies, Inc. VT8233/A/8235/8237 AC97 Audio Controller (rev 50)
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] K8 [Athlon64/Opteron] HyperTransport Technology Configuration
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] K8 [Athlon64/Opteron] Address Map
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] K8 [Athlon64/Opteron] DRAM Controller
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] K8 [Athlon64/Opteron] Miscellaneous Control
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] RV350/M10 [Mobility Radeon 9600 PRO Turbo]

Alsa init just returns this:
Lakka:/ # alsactl init
alsactl: sysfs_init:48: sysfs path '/sys' is invalid
alsactl: init:1758: No soundcards found...

I would be grateful is anyone would be able to help me with this. Thanks in advance.

Kivutar, 2015-02-23:
Post the output of aplay -L please

delta360, 2015-02-24:
Lakka:~ # aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)

Kivutar, 2015-02-24:
OK, if OpenELEC fixed it, let me upload an updated build...

Kivutar, 2015-02-24:
I uploaded a new build for PC here http://sources.lakka.tv/nightly/
Could you try it and tell me if this problem get fixed?
If not, I will merge openelec master again this week

delta360, 2015-02-24:
The problem is still there. Sorry.

delta360, 2015-02-24:
Looks as if I'm misinformed, newest OpenElec has the same problem. I'll contact them and get back to you.

delta360, 2015-02-25:
OK, guys from OpenElec told me that the chipset isn't supported, but I can try to compile myself.

https://github.com/OpenELEC/OpenELEC.tv/blob/master/projects/Generic/linux/linux.x86_64.conf#L3047

Would you be able to add support for this into Lakka if possible, if not I'll just try to compile myself. Thanks in advance.

Kivutar, 2015-02-25:
OK I just uploaded a nightly with this SND_VIA82XX compiled as a module. Please try it.

delta360, 2015-02-25:
Yes, it works perfectly, thank you!

closed 2015-02-25 by delta360