libretro/Lakka #182: Launching a ROM on RPi2 is slow

Kivutar, 2015-02-25:
It takes more than 30 seconds, sometimes 1 minute, to launch a ROM on the RPi2.

Does not affect RPi1.

You can reproduce by launching content from command line, or history, or browsing.

The log look like that

    Lakka:~ # retroarch -L /usr/lib/libretro/picodrive_libretro.so roms/sms/Alex\ Kidd.sms 
    RetroArch: parse_config_file: Loading config from: /storage/.config/retroarch/retroarch.cfg.
    RetroArch: load_symbols: Loading dynamic libretro from: "/usr/lib/libretro/picodrive_libretro.so"
    RetroArch: rarch_environment_cb: Environ SET_VARIABLES.
    RetroArch: verify_api_version: Version of libretro API: 1
    RetroArch: verify_api_version: Compiled against API: 1
    RetroArch: rarch_environment_cb: Environ PERFORMANCE_LEVEL: 0.
    RetroArch: rarch_environment_cb: Environ GET_LOG_INTERFACE.
    RetroArch: rarch_environment_cb: Environ SET_DISK_CONTROL_INTERFACE.
    RetroArch: rarch_environment_cb: Environ GET_VARIABLE picodrive_input1:
    RetroArch: rarch_environment_cb: 	3 button pad
    RetroArch: rarch_environment_cb: Environ GET_VARIABLE picodrive_input2:
    RetroArch: rarch_environment_cb: 	3 button pad
    RetroArch: rarch_environment_cb: Environ GET_VARIABLE picodrive_sprlim:
    RetroArch: rarch_environment_cb: 	disabled
    RetroArch: rarch_environment_cb: Environ GET_VARIABLE picodrive_ramcart:
    RetroArch: rarch_environment_cb: 	disabled
    RetroArch: rarch_environment_cb: Environ GET_VARIABLE picodrive_drc:
    RetroArch: rarch_environment_cb: 	enabled
    RetroArch: load_content: Content loading skipped. Implementation will load it on its own.
    RetroArch: rarch_environment_cb: Environ SET_PIXEL_FORMAT: RGB565.
    RetroArch: rarch_environment_cb: Environ SYSTEM_DIRECTORY: "/storage/system".
    RetroArch: init_content: Skipping SRAM load.
    RetroArch: audio_monitor_adjust_system_rates: Set audio input rate to: 44063.25 Hz.
    RetroArch: init_video: Video @ fullscreen
    RetroArch: init_video: Starting threaded video driver ...
    RetroArch: gfx_ctx_vc_init: [VC/EGL]: Initializing...
    RetroArch: gl_init: Found GL context: videocore
    RetroArch: gl_init: Detecting screen resolution 1920x1080.
    RetroArch: gl_init: [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
    RetroArch: gl_init: [GL]: Version: OpenGL ES 2.0.
    RetroArch: gl_query_extension: Querying GL extension: BGRA8888 => exists
    RetroArch: resolve_extensions: [GL]: BGRA8888 extension found for GLES.
    RetroArch: gl_query_extension: Querying GL extension: GL_EXT_unpack_subimage => doesn't exist
    RetroArch: gl_query_extension: Querying GL extension: EXT_sRGB => doesn't exist
    RetroArch: gl_init: GL: Using resolution 1920x1080
    RetroArch: gl_init: [GL]: Default shader backend found: glsl.
    RetroArch: gl_shader_init: [GL]: Using GLSL shader backend.
    RetroArch [WARN] :: gl_glsl_init :: [GL]: Stock GLSL shaders will be used.
    RetroArch: compile_program: Found GLSL vertex shader.
    RetroArch: print_shader_log: Shader log: Compiled
    RetroArch: compile_program: Found GLSL fragment shader.
    RetroArch: print_shader_log: Shader log: Compiled
    RetroArch: compile_program: Linking GLSL program.
    RetroArch: compile_program: Found GLSL vertex shader.
    RetroArch: print_shader_log: Shader log: Compiled
    RetroArch: compile_program: Found GLSL fragment shader.
    RetroArch: print_shader_log: Shader log: Compiled
    RetroArch: compile_program: Linking GLSL program.
    RetroArch: compile_program: Found GLSL vertex shader.
    RetroArch: print_shader_log: Shader log: Compiled
    RetroArch: compile_program: Found GLSL fragment shader.
    RetroArch: print_shader_log: Shader log: Compiled
    RetroArch: compile_program: Linking GLSL program.
    RetroArch: gl_init: [GL]: Using 4 textures.
    RetroArch: gl_init: [GL]: Loaded 1 program(s).
    RetroArch: gfx_ctx_vc_egl_query_extension: Querying EGL extension: KHR_image => exists
    RetroArch: font_renderer_create_default: Using font rendering backend: freetype.
    RetroArch: init_video_input: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
    RetroArch [WARN] :: udev_input_init :: [udev]: Couldn't open any keyboard, mouse or touchpad. Are permissions set correctly for /dev/input/event*?
    RetroArch: add_pad: [udev]: Plugged pad: Microsoft X-Box 360 pad (045e:028e) on port #0.
    RetroArch: input_try_autoconfigure_joypad_from_conf: Joypad port #0 (Microsoft X-Box 360 pad) configured.
    RetroArch: add_pad: [udev]: Pad #0 (/dev/input/event0) supports force feedback.
    RetroArch: add_pad: [udev]: Pad #0 (/dev/input/event0) supports 16 force feedback effects.
    RetroArch: input_joypad_init_driver: Found joypad driver: "udev".
    RetroArch: rarch_get_cpu_features: [CPUID]: Features: NEON
    RetroArch: alsathread_find_float_format: ALSA: Using signed 16-bit format.
    RetroArch: alsa_thread_init: ALSA: Period size: 384 frames
    RetroArch: alsa_thread_init: ALSA: Buffer size: 1536 frames
    RetroArch: rarch_get_cpu_features: [CPUID]: Features: NEON
    RetroArch: font_renderer_create_default: Using font rendering backend: freetype.
    <--- 30 sec delay here
    RetroArch: rarch_main_command: Loading history file: [/storage/.config/retroarch/retroarch-content-history.txt].

Note : launching RA without content is also kind of slow.

Kivutar, 2015-03-06:
Affects only lakka.c and xmb.c, not rgui

ochief, 2015-03-09:
I am sure this is obvious but the GLUI menu is also working as intended and much faster than lakka.c and xmb.c .

Kivutar, 2015-03-09:
With the old RetroArch shiped in Lakka, the bug affects only lakka.c. rgui, glui and xmb can launch roms fast.

With the latest RetroArch master, the bug affects xmb.c. Not rgui or glui. And lakka.c is gone anyway.

Kivutar, 2015-03-26:
I found a fix for this, it was due a texture format. The latest nightly builds contains this fix, so I can close this.

closed 2015-03-26 by Kivutar

Mwea, 2016-09-22:
I'm experiencing the same problem on PC. Can take two / three minutes to launch a ROM