libretro/Lakka #184: Multitap devices (NES, SNES)

rnavarropiris, 2015-03-02:
There is no option 'Multitap' for 'Device type' to enable 3+ players on NES (e.g Nintendo World Cup) or SNES (e.g. Super Bomberman 5)

Kivutar, 2015-03-02:
Hello, I was not able to use multitap on any SNES libretro core for a while, even using RGUI.
Could you declare this bug upstream, in RetroArch, please?

rnavarropiris, 2015-03-02:
@Kivutar Wasn't Snes9x-next integrated in the latest image? (Raspberry Pi 2). You also mentioned you successfully played Bomberman 5 on it in [this RetroArch issue]( https://github.com/libretro/RetroArch/issues/789)

Kivutar, 2015-03-02:
snes9x-next is on the latest image for RPi2 iirc. And this report about the multitap working was old. But I will try again today.

Kivutar, 2015-03-06:
I tried again today.

In input settings, I tried to set the port 2 as "SNES Multitap" and Bomberman 5 recognized it (you can set 5 "MAN" in a battle, MAN being a real player). However, I was not able to set more "SNES Joypad" after that. So I ended with only 2 controllers working...

rnavarropiris, 2015-03-06:
@Kivutar: Do you think that this is an issue in the [snes9x-next core](https://github.com/libretro/snes9x-next), in [retroarch](https://github.com/libretro/retroarch) or rather in Lakka?

I would like to post an issue in the correct platform(s), but I'm not familiar with the overall architecture (Lakka -> RetroArch -> snes9x-next). What do you think would be the best course of action?

Kivutar, 2015-12-03:
More information https://www.raspberrypi.org/forums/viewtopic.php?t=31681&p=747137

Kivutar, 2015-12-03:
So this is an issue with snes9x-next and bomberman5. I was able to play with 3 players on bomberman3. And snes9x+bomberman5 works will with 3 players.

Kivutar, 2015-12-07:
I declared the new issue here, since it's where it belongs https://github.com/libretro/snes9x-next/issues/63

closed 2015-12-07 by Kivutar