libretro/Lakka #186: [Enhancent] Integration of Limelight Streamer

jayp76, 2015-03-07:
Integration of Limelight Streamer

https://github.com/irtimmer/limelight-embedded

Atarian2600, 2015-03-17:
I'd like to see this, too.

Kivutar, 2015-03-17:
You can request a port of a limelight client to libretro... but it's quite a lot of work, since none of their client seems to be written in C. And none of them seems to be portable easilly, since they wrote independant code for each mobile platforms.

Atarian2600, 2015-03-17:
I seem to remember people having limited success getting limelight working in OpenELEC. That was roughly a year ago, I believe. I wonder if any progress was made with that.

Kivutar, 2015-03-17:
This limelight embedded, is it C or Java? github sais it uses both.

Kivutar, 2015-03-17:
That is java, we can't integrate it in RetroArch.

stsfin, 2015-03-31:
Hi, I made scripts for integrating limelight with emulation station/retropie, I hope you can fork these for Lakka: https://github.com/stsfin/RetropieLimelightInstaller

Here is just a generic installer/setting reconfigurator for rasbian based systems: https://github.com/stsfin/LimelightPiInstaller

sigboe, 2015-04-12:
@Atarian2600 Right now you would need to have Oracle Java installed as well as a shared Object libopus. If you manage to install these on OpenElec, limelight should be startable.

@stsfin Openelec does not have aptitude or dpkg, and the main partition is not persistent after boot. OpenELEC is not based on rasbian.

stsfin, 2015-04-12:
Hi,

Here are is a OpenElec w/ Kodi guide:
https://github.com/HazCod/LimeRPi2-kodi/blob/master/README.md

richardwolfe4991, 2015-04-17:
So has anyone got this working? I'd love for this to be a reality

Kivutar, 2015-04-19:
Just to let you know, RetroArch already include a video recording / streaming solution, with ffmpeg. I disabled it for now, but I could try to support it.
I noticed that they have been adding some menu interface for this functionnality.
It can stream to twitch, and certainly elsewhere. I never tried it yet.

fr500, 2015-04-19:
A limelight "player" core would be amazing, most of the backend code seems to be C so it should be doable

Kivutar, 2015-04-19:
@fr500 I agree