libretro/Lakka #187: [Enhancement] Use the new Libretro-FBA core

ochief, 2015-03-08:
Could you please add the https://github.com/libretro/libretro-fba core.  I do not believe core's can be compiled manually.

I believe the makefile.libretro can support multiple platforms and also has a PROFILE=Performance for low powered platforms such as the Raspberry Pi.

Thanks

Kivutar, 2015-03-08:
Done. I tryed the profiles "performance" and "accuracy" and noticed no difference. Also, I noticed no performance drop betweeen the old and the new versions.

Tested with Demon Front on RPi2.

closed 2015-03-08 by Kivutar

ochief, 2015-03-09:
Thanks for adding this so quickly.

The "performance" profile uses a different version of the SH-2 CPU core which wouldn't have effected Demon Front.  It is however used for CPS-3 games.

Here is a list of parents/clones that would be effected.  Should be noticeably quicker for RPI, Odroid etc.

http://www.mamedb.com/cpu/SH-2

Kivutar, 2015-03-09:
Should I use "performance" for all our ARM projects, and "accuracy" for PC?

ochief, 2015-03-09:
I'm not an expert by any means but I think that would be a safe bet.

BarbuDreadMon who created the libretro-fba core found that the newer version of the SH-2 core used in mainline FBA dropped performance from 60fps to 45fps for CPS-3 games on his Arm device (I think a RPI2).

http://libretro.com/forums/showthread.php?t=2120&page=2

I think it would be a good idea for all ARM projects to use the older SH-2 core for now.  The version used in the "Performance" profile is the same version that was used in the previous FBA core (libretro-fba) used by Lakka.

On my RPI2 I was comfortably getting 60fps on Street Fighter 3 with this core.

Kivutar, 2015-03-09:
OK, let's do that for now 9462145ed7ff8e3dea84c48c76426e475f2de56f