libretro/Lakka #191: [Enhancement] - Move settings to the end of the list

ochief, 2015-03-19:
Something like this may already be in the pipeline so this may be unnecessary.

At the moment when you load Lakka, the first list you see is the settings options which is useful when first configuring Retroarch but once this is done you are unlikely to need it again (or not regularly).  It would make more sense to move it to the end of the lists, after all of the cores that are displayed.

It would be nice if you could rearrange the order for everything so it didn't show the cores in alphabetical order.  Theoretically you could change it to the following (if you desired).  That would also mean that you can hide unwanted ones by simply not selecting an option.

Menu 1 - Snes
Menu 2 - Settings
Menu 3 - Arcade
Menu 4 - Coleco
Menu 5 - None
Menu 6 - None
etc....

Moving the settings menu to the end may be a simpler option if you are not going to be re-working the menu system heavily.

projectb-temp, 2015-04-10:
I second this FR partly.
But instead of moving it completely to the end maybe this could be an option:
Menu 0 - Reboot Lakka / Shutdown Lakka / Restart Retroarch
Menu 1 - Core 1
Menu 2 - Core 2
Menu 3 - Core 3
Menu 5 - Core n
Menu n - Settings

Additionally enabling the option for the menu to wrap-around in a circle could enable the user to go from Menu 0 (Reboot...) easily to Menu n (Setings) with just one click.

tiagoanleite, 2015-05-15:
The first setting on my installation is Settings, either with Lakka and XMB menu drivers.
(Using the release available on the website today)

Kivutar, 2015-11-20:
There is an horizontal wraparound option in the settings now, and XMB honor it.
Also, the new scanning system reduces dramatically the number of playlists in the horizontal menu, since you will just add the systems you play. So I close this for now. Adding icons at the end of the list is difficult for programmatic reasons :(

closed 2015-11-20 by Kivutar