libretro/Lakka #192: View retroarch debugging output

garnieretienne, 2015-03-28:
I'm trying to have mupen64plus core working on my build (a Generic x64 system) (see https://github.com/libretro/Lakka/issues/138) and was wondering if lakka have a way to see the retroarch logging info to see what append. 

I've tried to change the `ExecStart=/usr/bin/retroarch` line to `ExecStart=/usr/bin/retroarch &> /storage/retroarch.log` in the `retroarch.service` file and rebuilding the package but lakka refuse to start with this config (I'n not familiar with systemd).

Kivutar, 2015-03-28:
Hello, you can loggin using ssh, and stop the retroarch service, then launch it by hand, and you will get the output in the stdout.

garnieretienne, 2015-03-29:
If other search for this, I figured out I can use the `journalctl` command with ssh to follow retroarch output:
```console
journalctl --unit retroarch --follow
```

closed 2015-03-29 by garnieretienne

infotek, 2016-05-20:
I used
systemctl stop retroarch.service ; retroarch -v