libretro/Lakka #194: Set resolution on Lakka

jago984, 2015-04-03:
Hi! I have a PC with VGA connection on a 40" Monitor.
It detects automatically the 1920x1080 resolution but it's working slow with some emulator so I wanted to force a different resolution but nothing seems to work.
I tried editing manually retroarch.cfg with:

video_fullscreen_x = "1280"
video_fullscreen_y = "800"

But as I restart retroarch it brings it back to the default:

video_fullscreen_x = "0"
video_fullscreen_y = "0"

Just as troubleshooting I will try later with a smallest monitor to check the frame rate.

floreal, 2015-04-03:
Hello,
Before editing your config file, ensure you stopped retroarch service:

systemctl stop Retroarch

Then start it again after saving your modifications.
By default retraarch rewrites the configuration file when it exits on lakka.

jago984, 2015-04-03:
Thank you very much, it was simplier than I expected, I managed to change the resolution but I still have very slow framerate (40fps and buggy sounds with psx games)... It's strange since this pc is not so old and I've tried Lakka on an older notebook without issues with the same games.

jago984, 2015-04-03:
UPDATE: I tried with snes and sega ms, it works fine with sega ms but it's still buggy with snes... it's a 64 bit processor with 4 gb ram... seems strange

floreal, 2015-04-03:
What is your GPU ?

jago984, 2015-04-03:
NVIDIA® ION™(GT218-ION™) the pc is a Asus EeeBox PC EB1501P with Intel Atom dual core.

jago984, 2015-04-03:
Now I got it working with snes9x too, i checked here: https://github.com/libretro/RetroArch/wiki/Getting-optimal-vsync-performance to verify vsync, but psx is still slow.
It gives me continuosly: page_flip_handler: [KMS/EGL]: Missed 1 VBlank(s)
RetroArch: rarch_log_libretro: [libretro INFO] :: [Beetle PSX]: Samples / Frame: 737.19316

The end of the log is:

RetroArch: rarch_log_libretro: [libretro INFO] :: [Beetle PSX]: Estimated FPS: 59.82150
RetroArch: compute_audio_buffer_statistics: Average audio buffer saturation: 2.58 %, standard deviation (percentage points): 10.73 %.
RetroArch: compute_audio_buffer_statistics: Amount of time spent close to underrun: 95.21 %. Close to blocking: 0.34 %.
RetroArch: page_flip_handler: [KMS/EGL]: Missed 1 VBlank(s) (Frame: 1063, DRM frame: 284624).
RetroArch: thread_free: Threaded video stats: Frames pushed: 565, Frames dropped: 20.

Kivutar, 2015-04-06:
Maybe the nouveau driver does not work well on ION?

Yqnn, 2015-05-18:
Hello,
I've exactly the same issue with the same GPU (ION GT218), on a different machine : Asus S1-AT5NM10E.
Is there a way to try the NVIDIA proprietary driver instead of Nouveau ?

Kivutar, 2015-05-19:
This image have the proprietary nvidia driver, please tell me if it works http://static.kivutar.me/Lakka-Generic.x86_64-devel-20150518150748-r21009-gd55ab55.img.gz

zaidmian, 2015-05-20:
Any chance of getting the tar for 21009? I'd like to update in-place.

Kivutar, 2015-05-21:
@zaidmian you can mount the img with daemontools or mount loop or bsdtar and get the KERNEL and SYSTEM files and override the ones of your system manually.

zaidmian, 2015-05-21:
I'm not sure what the problem is but I can't seem to mount that img no matter what I try. Maybe I have a corrupt download? I'll try again.

zaidmian, 2015-05-21:
@Kivutar I'm on Windows btw.

zaidmian, 2015-05-21:
@Kivutar Sorry for the spam. I got it. Thanks.

Yqnn, 2015-05-21:
Thank you very much for the image. 

I tried it yesterday, and unfortunately  it seems a bit slower than the one using Nouveau drivers.

I checked the CPU usage, and Retroarch is using one core at 100% (among the 4 virtual cores of my CPU).
I guess it could also be a CPU issue, maybe Beetle PSX is too demanding for a simple Atom ?

Kivutar, 2015-05-21:
I think nouveau+kms is better at blitting the whole image, pcsx_rearmed being software rendered, you will get better performances for it. 

However, if you try a libretro GL core, like PPSSPP, the proprietary driver should be way faster.

matcarfer, 2015-10-19:
I'm sorry but the reason emulation is slow is that CPU is an Atom D525, and its slow as hell:

https://www.cpubenchmark.net/cpu.php?cpu=Intel+Atom+D525+%40+1.80GHz

Yqnn, 2015-11-19:
True, the Atom D525 is very slow.

But the PSX CPU is only 33 MHz, and I remember that Virtual Game Station was able to run PSX games at full speed on a 500 Mhz G3 (more or less 10 times slower than the D525).
I believe VGS was aggressively optimized, but It's a pity to require such a powerfull CPU to emulate a 20 years old console.

matcarfer, 2015-11-20:
VGS had A LOT of optimizations and hacks to improve emulation speed. Mednafen aims to emulate the best it can, two different aproachs. As it is, it consumes more CPU power (as BSNES compared to Snes9x Next). First Atom CPUs like yours sadly are really slow, so slow that it compared to old Pentium 3 procesors. Try Beetle PSX core, maybe you will have better luck.

closed 2015-12-29 by Kivutar