libretro/Lakka #196: Dualshock wireless won't autoconfigure

JamieMagee, 2015-04-16:
I am running the latest nightly on my RPi2 and I have configure bluetooth using the instructions in the wiki. My two dualshock controllers are detected and autoconfigured when connected through USB, and they are correctly detected as dualshock controllers over USB, but the controls are not mapped correctly. The only buttons that seem to work are L2 and R3 (and not with the correct functions)

Kivutar, 2015-04-18:
Yes, it's a known issue. The wireless dualshock 3 has the same name as the wired duarlshock 4. The wired dualshock 4 config file. So we never added a wireless dualshock 3 config file. It would conflict.

There is no much thing we can do for now. Except maybe if we build a bluetooth stack into retroarch and find a better way to match the files.

JamieMagee, 2015-04-18:
Okay, but can I overwrite the dualshock 4 wireless config file with the dualshock 3 config file on my installation? Or is that read only?

Kivutar, 2015-04-19:
You can override the configuration. Many users got their DS3 wireless working on Lakka. You can either override that file (by overriding the path where those config files are stored, make it point somewhere in your readwrite partition) or by using the input settings gui.

Kivutar, 2015-05-12:
Some additional info: with a 4.0 kernel, the wireless DS3 has the right name, and is binded correctly.

JamieMagee, 2015-05-13:
So Lakka will support this when OpenELEC 6.0 is merged? When are you expecting to do that?

Kivutar, 2015-05-13:
I could do it now, but OpenELEC is droping i386 and BIOS support with the 6.0, and I don't want to lose those features. Keeping i386 can be easy, but BIOS is not.

While we think about a solution for this problem, we could use a quick workaround to support DS3 wireless, but it will break DS4 wired and wireless support. I don't know which one we should support in priority.

ochief, 2015-05-18:
If you do add a work around pleaseplease make it optional.  I have just brough 3 ds4 controllers and would be gutted if I can't use them with Lakka Going forward.

Kivutar, 2015-05-19:
Don't worry for this. When we will upgrade to linux 4, the bug will be fixed for everybody, there will not be any regression on this.

Kivutar, 2015-06-25:
The DS3 I bought is recognized as https://github.com/libretro/retroarch-joypad-autoconfig/blob/master/udev/Sony_PLAYSTATION%28R%293_Controller.cfg with bluetooth and autoconfig works out of the box for it.

Also, on the forums, another user got it autoconfiguring with bluetooth without problems: http://libretro.com/forums/showthread.php?t=3167

Kivutar, 2015-08-06:
@JamieMagee the wiki have been updated with informations about how to recognize (very good) fakes. http://www.lakka.tv/doc/Wireless-Dualshock/

Can you check your controller?

JamieMagee, 2015-08-11:
My Pi is currently re-purposed for something else, so I can't check. I got one of the controllers with a PS3, and the other I bought in a GameStop. I assume at least one of them is genuine.

Kivutar, 2015-11-20:
Any news from this?

chuckwired, 2015-12-06:
If it helps, I've tested two DS3 controllers wirelessly at the same time with a nightly build from 19th November 2015 and can confirm they work flawlessly.

Kivutar, 2015-12-07:
Thanks, I can close this.

closed 2015-12-07 by Kivutar