libretro/Lakka #200: Audio not working on TV

aldatsa, 2015-05-01:
My TV didn't play any sound when using Lakka.

I found that adding

>hdmi_drive=2

 to the config.txt as suggested in #33 fixes the issue.

I think that this was fixed in #29 but that line was removed later and is not present in the current release.

Kivutar, 2015-05-02:
Yes some TV send an EDID saying that they don't support HDMI audio while they do support HDMI audio. This cause the RPi bootloader to disable HDMI audio and you get no audio on those TVs.

There are two problems when you set hdmi_drive=2:

 * Jack audio will stop to work
 * Using an HDMI to DVI converter, your Pi will just not boot

I thought at that time that we should let the bootloader config as is, since the bug here comes from the TV and we can't get this config file to work in all cases at the same time.

What do you think @floreal ?

aldatsa, 2015-05-02:
How about adding something like this to config.txt:

   
################################################################################
    # If you don't get sound on a TV connected by HDMI to your Raspberry Pi
    # try with this setting uncommented.
    # https://github\.com/libretro/Lakka/issues/200
    ################################################################################
    # hdmi_drive=2

Kivutar, 2015-05-02:
Yes, this would be perfect

aldatsa, 2015-05-02:
Ok. I updated my pull request (#201) to reflect these changes.

PR #201 (2015-05-02): Fix HDMI sound on some TVs
Fixes #200
closed 2015-05-02 by Kivutar