libretro/Lakka #204: Translations

Airon90, 2015-05-15:
Is it possible to make Lakka translateable? Thank you very much!

Kivutar, 2015-05-15:
There was a tentative to translate RetroArch at some point, but there are some issues about it:
 - gettext is not so portable, and RetroArch need to run on so many platforms
 - RetroArch evolves so fast, it would be difficult to keep the translations up-to-date

I think it's not the right moment to implement translations in RA. We should think about when the development calms down a little (if it ever happens).

Airon90, 2015-05-15:
Ok, thank you very much for your answer!
Please advise me when you start translating Lakka: translating is my way to contribute to projects as I don't know how to code :)

Kivutar, 2015-06-03:
I close this for now.
Translation is something that belongs to RetroArch, not Lakka.
And it's very complex to implement because of our constraints.

Kivutar, 2015-06-20:
Hello @Airon90 , a translation system is being added to RetroArch. You talk italian right?

Airon90, 2015-06-20:
I can translate into Italian (it), as mothertongue language, and Esperanto (eo), as advanced translator.

Kivutar, 2015-06-25:
The file for italian is in RetroArch repo, under menu/intl/menu_hash_it.c
The strings that need to be translated are those beginning by MENU_VALUE_ and MENU_LABEL_VALUE_. But strings beginning just by MENU_LABEL_ should be excluded.

We are also interested in the Esperanto translation.

Kivutar, 2015-06-26:
The file for esperanto have been added too.

closed 2015-07-14 by Kivutar

Airon90, 2015-07-26:
Thank you but I didn't find any "menu" directory!

Kivutar, 2015-07-26:
Ah, the italian file is here https://github.com/libretro/RetroArch/blob/master/menu/intl/menu_hash_it.c
The english file is here https://github.com/libretro/RetroArch/blob/master/menu/intl/menu_hash_us.c
You can find the other files here https://github.com/libretro/RetroArch/tree/master/menu/intl