libretro/Lakka #207: "Black Screen" with Genesis Emulator on x86_64 port

jayp76, 2015-05-16:
Hi,

on latest nightly Lakka-Generic.x86_64-devel-20150512171823-r21002-ga5310d6 there is a "black" screen with the Genesis Emulator. Either it is a black screen or it is a 1 pixel wide screen. Sound is OK. 

Have a nice weekend.

jayp76, 2015-05-16:
Bug also on Rpi2 Port.
Also N64 Emulator affected.

Kivutar, 2015-05-19:
I also noticed this bug on genesis-plus-gx.

However, the N64 bug is a separate bug, it's just caused by the active development on this core, and should be fixed soon.

twinaphex, 2015-05-19:
Have him retest N64 again. Over the past three days there have been some GLES2 regression fixes by heuripedes.

Kivutar, 2015-05-19:
I'm upgrading that package right now to test. I saw the log about the fix yes.

jayp76, 2015-05-26:
I tested now the latest nightly from 26th of May and it got worse. All Emulators video output is 1 pixel wide. I found out that under Video Preferences the Pixel Aspect Ratio Index is "Custom". When i choose "Core provided " all Emulators are fine again. Unfortunately when i reboot, the Pixel Aspect Ratio Index is again "Custom".

twinaphex, 2015-05-26:
You can reconfigure the 'custom aspect ratio' by going to Options -> Video Options, Selecting 'Custom Ratio' and setting the dimensions of the screen.

jayp76, 2015-05-28:
It eventually save the pixel aspect ratio parameter after several tries. i couldn't find a pattern when it saves and when not. seems wonky to me.

twinaphex, 2015-05-28:
I don't understand what you're saying, the way Custom Aspect Ratio works is that you (obviously) set the parameters per corner of the screen, then after you've input all that, the custom aspect ratio viewport settings are saved and you can toggle them by going to Aspect Ratio Index and selecting 'Custom'.

Honestly I just think you're using it wrong.

jayp76, 2015-05-28:
What i meant is that at some point the configuration seemed to be faulty. I discovered that the apsect pixel ratio setting was set to "Custom". That is causing that behaviour i mentioned earlier with a 1 pixel wide video output. When it is set to "Core provided" this behaviour is gone and video output is normal. But that setting was not permanent when i restarted retroarch or rebooted. The setting saving mechanism is not reliable in certain cases it seems.

twinaphex, 2015-05-28:
It is reliable, it is just that RetroArch only saves the config at exit.

Kivutar, 2015-05-29:
@jayp76 can you try to set the value to "core provided" and restart or quit retroarch via the menu?

This will save the config.

jayp76, 2015-05-29:
As i mentioned, i tried this and it worked at last, but not every time i tried it (quitting or restarting retroarch).
For now case can be closed.

Kivutar, 2015-05-29:
OK, you can still reopen it if you still have the bug or learn more about the cause.

closed 2015-05-29 by Kivutar