libretro/Lakka #208: loud fans on XMB menu

ochief, 2015-05-19:
When I load Lakka and get to the main menu screen my fan is running much faster than it needs to.  As soon as I load a game it cuts the speed and noise down considerably.

Is this something you are aware of/expect and assuming not what could I do to help troubleshoot this?  Is there an obvious command to see how much CPU Lakka is using?

Kivutar, 2015-05-20:
This bug was supposed to be fixed by the framebuf.dirty optimization by @enygmata.

You can ssh and use top?

ochief, 2015-10-02:
[image]

Sorry for the delay.  Is there anything else I could look at, it doesn't seem that it is using a lot of CPU at this stage?  This is using a recent build of Lakka with X enabled.  I can try again with the recent KMS version you posted but is it worth me running any more tests with my current setup?

Kivutar, 2015-10-03:
I would be curious to know if the same problem happens in the KMS version yes.

Can you try with RGUI and GLUI too?

ochief, 2015-10-05:
Running it with KMS doesn't cause any issues with loud fans.

I will go back to the previous build and see if there is a difference between the menu drivers and post my results.

Kivutar, 2015-11-30:
This should be fixed now. The runloop had many fixes recently. If you're not using extensively the features like boxarts, the cpu activity should stay low.

I close this for now. If it still happens, please reopen it!

closed 2015-11-30 by Kivutar