libretro/Lakka #209: Stuck at blank screen after the flower screen on a clean install (same from Live boot)

lispington, 2015-05-20:
Hi a fresh Lakka install on an old Acer Aspire 5630 laptop (3gb mem).

1. After install (with and without SSH) the flower screen appears, then goes, and the HDD light flickers at about 5 second intervals, constantly (I left for about an hour once - just to see).
2. Live boot goes through the initial creation of space, then after its follow on boot exibits the same behaviour.
3. Once booted (and a blank screen), I can still connect as root to the machine via SSH. I can browse the directory structure, perform tasks like lsusb etc (it spots my PS3 controller by the way). And copy files to the roms directory. But still no actual action on the laptop.
4. Looked at the BIOS (old) and there are no options to set/unset any of the weirder settings in newer BIOSes I have seen.
5. The version (image)  I am installing is : Lakka-Generic.i386-devel-20150307205832-r20834-ge8d52d6.img.gz

(I also tried one version from the nightlys, Lakka-Generic.i386-devel-20150504114310-r21001-g922391b, but same problem)

If you want me to add some log files I can get via SSH or perform some tests I can.

Thanks

Kivutar, 2015-05-21:
Thanks for your report. Can you confirm that the graphical card is a GMA 950?

lispington, 2015-05-21:
lspci reports :

VGA compatible controller: NVIDIA Corporation G72M [Quadro NVS 110M/GeForce Go 7300] (rev a1)

Kivutar, 2015-05-21:
Can you try this image instead? http://static.kivutar.me/Lakka-Generic.x86_64-devel-20150518150748-r21009-gd55ab55.img.gz

It uses X11 for rendering instead of drm/kms.

lispington, 2015-05-21:
And here is lspci -v for the techy stuff! 

VGA compatible controller: NVIDIA Corporation G72M [Quadro NVS 110M/GeForce Go 7300] [rev a1] (prog-if 00 [VGA controller])
        Subsystem: Acer Incorporated [ALI] G72M [Quadro NVS 110M/GeForce Go 7300]
        Flags: bus master, fast devsel, latency 0, IRQ 29
        Memory at cd000000 (32-bit, non-prefetchable) [size=16M]
        Memory at d0000000 (64-bit, prefetchable) [size=256M]
        Memory at ce000000 (64-bit, non-prefetchable) [size=16M]
        Expansion ROM at <unassigned> [disabled]
        Capabilities: [60] Power Management version 2
        Capabilities: [68] MSI: Enable+ Count=1/1 Maskable- 64bit+
        Capabilities: [78] Express Endpoint, MSI 00
        Capabilities: [100] Virtual Channel
        Capabilities: [128] Power Budgeting <?>
        Kernel driver in use: nouveau

lispington, 2015-05-21:
It's a 32bit system, but I'll try this.

Kivutar, 2015-05-21:
ah, wait, will not work on a 32bit system. I will upload a new image asap.

lispington, 2015-05-21:
Sorry. I am not sure anymore. I thought this old thing was 32bit, now I google it, it could be 64bit. Anyway, it installed the image and just loops round with the message:

"failed to start xorg.
is your gpu supported?"

looking at dmesg | grep NVRM the following is displayed :

[    7.149486] NVRM: The NVIDIA GeForce Go 7300 GPU installed in this system is
[    7.149486] NVRM:  supported through the NVIDIA 304.xx Legacy drivers. Please
[    7.149486] NVRM:  visit http://www.nvidia.com/object/unix.html for more
[    7.149486] NVRM:  information.  The 340.76 NVIDIA driver will ignore
[    7.149486] NVRM:  this GPU.  Continuing probe...
[    7.149504] NVRM: No NVIDIA graphics adapter found!
[    7.149615] NVRM: NVIDIA init module failed!

Kivutar, 2015-05-21:
OK, so your gpu is old, and need the old driver.

There is a new project called nvidia-legacy in OpenELEC that addresses this problem. However, we don't want to merge OpenELEC right now because they are droping BIOS and i386 support.

lispington, 2015-05-21:
I see, well thanks anyway. Very grateful for your help. At least I know the Acer is 64bit now, and I can find another machine to install Lakka on.

closed 2015-05-21 by lispington

Kivutar, 2015-05-21:
Not problem. If I manage to merge OpenELEC while keeping BIOS and i386, I will let it know on the blog, and will release for nvidia-legacy.

ltclarke, 2017-05-19:
Hi, 

I am doing a new install with the same symptoms on an Acer Aspire One. I suspect it has the same issue with an old gpu. 

> If I manage to merge OpenELEC while keeping BIOS and i386, I will let it know on the blog, and will release for nvidia-legacy

Wondering if this ended up happening? Is there an alternate install version with this support included?

Thanks!