libretro/Lakka #210: i386 Fix on flower after installation

memorex37, 2015-05-21:
Hello,

I use a Giada i33 Processeur Intel Atom D525, and i have this problem :
After installation, on the startup, it's fix on flower ? The installation is OK. I use different nightly and it's the same trouble.

Have you some information about that ?

Expect of that, i install lubuntu and retroarch and all is OK, but without interface, it's not simple for kids.

Regard

kinta18, 2015-05-25:
Hi, i have an old pc and the installation is OK but when i try to boot it stays in the flower and nothing happens. Specs: Intel celeron D 3.3 ghz, motherboard P4M890-M7 SE Versión 7, 1 gb of ram, ide  HDD 80gb and 64 mb of video. in my new rig works awesome, but in my old computers dont, i was thinking in the processor instructions or something like that, but i cant figure it out.

Kivutar, 2015-05-25:
Hello, this can happen... As said on the hardware support page, there are so much PC out there, it's impossible to support them all.

The most common cause, is that we use some instruction sets not supported by your cpu.

But not using those instruction sets would mean making lakka slower for everybody else.

memorex37, 2015-05-26:
Thanks a lot.

closed 2015-05-26 by memorex37

Kivutar, 2015-05-26:
There is a group of people on the forums trying to compile Lakka with less instruction sets. If you have some skills about compiling, you can join them.