libretro/Lakka #212: Lakka reboot Loop

Halor69, 2015-06-05:
Not sure the exact version, but it is the last stable version before the June 2 released version.  I can get the version in a little bit when my youngest stops playing.  I was not in the room when it happened, but he stated that he pressed the "ESC" key on the keyboard twice and it went into a reload/reboot loop, flower splash, short flash then flower splash, over and over.  I tried to do just a repair with the installer, but that was a no-go.  So went and re-installed everything.  We didn't really have any save games to worry about this time, but might at a later date.  Is this a known issue?

Kivutar, 2015-06-06:
Hi, thanks for the report. No, it is not a know issue. Maybe a filesystem corruption.

Kivutar, 2015-11-30:
I was never able to reprodude the bug. In the hope that this bug is something exceptionnal that will not happen more, I close this report. Feel free to reopen it if it happens again.

closed 2015-11-30 by Kivutar