libretro/Lakka #214: Missing icons

Kivutar, 2015-06-24:
<Twinaphex> Famicom Disk System
<Twinaphex> Oddyssey 2
<Twinaphex> Gamepark GP32
<Twinaphex> Entex- Adventure Vision
<Twinaphex> ColecoVision
<Twinaphex> Casio - Loopy
<Twinaphex> Quake - RCA Studio II
<Twinaphex> Sega PICO
<Twinaphex> Tiger Game.com
<Twinaphex> VTech - CreatiVision
<Twinaphex> Watara Supervision

sergiobenrocha2, 2015-07-06:
It's missing MSX2, is it the same for MSX?

Kivutar, 2015-07-07:
We should have one icon per rdb, so if we have separate rdb for MSX and MSX2, then we should have 2 separate icons.

sergiobenrocha2, 2015-07-07:
What's the default or most used media for MSX? The wiki says some MSX models didn't come with floppy, maybe it make more sense use cartridge instead?
http://www.msx.org/wiki/MSX_for_beginners

Some cartridges:
http://www.grauw.nl/msx/imagegallery/

I liked this one:
[image]

sergiobenrocha2, 2015-07-07:
The others may be confused with genesis cartridges.

Kivutar, 2015-07-08:
Oh I like this one, it is easily recognizable. I have no idea what is the most common media for MSX though :(

sergiobenrocha2, 2015-07-08:
A already did a PR ;)

Kivutar, 2015-07-08:
Merged.

A user sent me a Game & Watch icon by mail. I had to fix it a little. Now the problem is I don't know how to name the file.

sergiobenrocha2, 2015-07-08:
It's missing History list icon yet.

sergiobenrocha2, 2015-07-08:
Btw, why mame's content icon is a coin (and I needed someone told me it was a coin lol :p )? Would not it be better to use an eletronic chip instead? Or 2 chips?

Kivutar, 2015-07-08:
Well, that was not an easy choice I admit... The rationnale was that most users ignore that games in arcade are stored on ship. They have never seen an arcade ship of their life. Only arcade specialists and emulator people know that. While the user have in his mind that one game session equal one coin.

kerunaru, 2016-02-16:
Just for the sake of knowledge… The most common media on MSX systems was the cartridge. :)

closed 2016-08-11 by Kivutar