libretro/Lakka #215: lakka x86 does not auto-mount USB

xmat123x, 2015-06-25:
Tried on 2 different machines. Changing the path by going into /storage/roms/ is empty even with a fat32 thumb drive connected.

This works perfectly on the Raspberry pi build.

Kivutar, 2015-06-25:
Yes, udevil has been disabled by default on Generic images. It caused some problems when testing lakka live on a machine with big drives full of data, with our automatic scanning.

Soon, the scanning will not be automatic more, and we will re-enable this feature on Generic.

xmat123x, 2015-06-25:
Thanks for replying so quickly. Is there anywhere I can download a build that has it enabled? I absolutely love lakka and wanted to install it on my netbook HDD as the only primary OS. I tried messing around with SSH to mount the drive manually but that didn't work too well.

Kivutar, 2015-06-25:
Oh, there is a topic on our forums where sombody used the autostart.sh script to automount his external disk automatically on boot. I think you should try that.

In the meanwhile, I will try to dev a temporary solution to enable/disable udevil from the bootloader. But it can cause problems, I can't promise at 100% that I will push it. Especially in this case, where we're going to have a clean fix soon.

xmat123x, 2015-06-25:
Ok, cool. I'll be monitoring the commits.

Kivutar, 2015-06-25:
I uploaded a new PC nightly with udevil enabled. I disabled udevil in the live usb mode. To avoid the problems I was talking about. But it should now work when Lakka is installed on a dedicated PC. Let me know if it works as expected, and I will push.

xmat123x, 2015-06-25:
Just tested it and everything worked beautifully. Lakka has given new life to my aging netbook!

Kivutar, 2015-06-25:
Good to hear. I will push that.

closed 2015-06-25 by Kivutar