libretro/Lakka #217: Suggestion for Wifi config documentation

leojimenezcr, 2015-06-29:
The wifi documentation need more information (http://www.lakka.tv/doc/Network-settings/). My suggestion:

## Wifi

For now, the graphical interface of RetroArch does not allow to change the network settings, you will have to use the [command line interface](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface).

Connecting to the network requires the *conmannctl* command.
First enable the wifi interface:
`# connmanctl enable wifi`

For protected access points start interactive mode. It will ask for the wifi password:
`# connmanctl agent on`

To scan the area for wifi networks run:
`# connmanctl scan wifi`

To list the wifi networks from the above scan run:
`# connmanctl services`
`android-device-tether wifi_de65fr56325_524632863278`
`FBIDrone wifi_je86fe48321_532486348931`

To connect to one of the wifi networks use the section starting with ‘wifi_’
`$ connmanctl connect wifi_de65fr56325_524632863278`

You can now be on the network. Check it out by using *ip addr* or *connmanctl state*.

Once done, your Lakka Box should automatically connect on startup. If not, you way have to mark the service as prefered.

Kivutar, 2015-06-29:
I agree, thanks. You can edit this page: https://github.com/libretro/Lakka/wiki/Network-settings
Our website will auto update after some time.

leojimenezcr, 2015-06-29:
I just updated the page. Done ;-)

closed 2015-06-29 by Kivutar

roba91, 2016-07-29:
Little hint in "If not, you way have to mark the service as prefered."

This is problematic as archwiki tells you you have to edit a config file in /etc/... but this folder is read only.

Please expand the documentation on how to exactly make a service prefered.

Kivutar, 2016-07-29:
I personally never had to mark a network as preferred. Are there really cases where it is needed?

leojimenezcr, 2016-07-29:
2016-07-29 14:24 GMT-06:00 Rouven Bauer <notifications@github.com>:

> Little hint in "If not, you way have to mark the service as prefered."
>
> This is problematic as archwiki tells you you have to edit a config file
> in /etc/... but this folder is read only.
>
> Please expand the documentation on how to exactly make a service prefered.
>

Hi,

I don't know why. I just keep the paragraph from original article.

*Leo JiménezSan José, Costa Rica*
* +506 8840 5095*  [image: @leojimenezcr] <https://twitter.com/leojimenezcr>
[image: Leo Jiménez] <https://facebook.com/leojimenezcr>  [image:
https://telegram.me/leojimenezcr] <https://telegram.me/leojimenezcr>

roba91, 2016-07-29:
Raspberry Pi 2 with USB Wifi card wants to use the ethernet connection even though there is no cable connected.

To answer @leojimenezcr 's "why": Because I have  no idea how to do this, if one would really want to do so (and yes i want). So option a) is to remove that hint or b) explain how exactly to do it as it's really not a trivial task. I think c) keeping it as is, is the worst option.

roba91, 2016-07-29:
Correction: Sorry it's just a RPi v1 B

WaifThe3rd, 2016-10-19:
Hello,

I'm running into some issues with connmanctl. I do all the steps in the guide while my box is connected via ethernet. Once connected I get an ip addr for the wireless and ethernet, and i can ssh to it via wireless ip, which is good. Once I reboot it and remove the ethernet cable, i never get a wifi ip automatically. Once I reconnect the ethernet cable I ssh into the box and do a connmanctl services it only shows the wired connection is connected *AR with no other options, after some time i ran connmanctl services again and it shows *A SSID, but I dont have an ip addr for the wireless. Then I tried to run connmanctl connect wifi_ (tab) and nothing happens, then i typed out the whole wifi_xyw01231 stuff and no connection. I then have to re-scan the wifi, and it will connect wifi_ and give me an ip addr.

I wanted to check on this preferred service if it will help but like they say the main.conf is read only and you cant chmod it.

Is there anyways to make this autoconnect on start up?

Kivutar, 2016-10-19:
Try to set the wifi network as preferred in connmanctl.

WaifThe3rd, 2016-10-19:
Is there a way to do this with a command? All I can find is to change the preferred status in the main.conf file for connmanctl.

Kivutar, 2016-10-19:
Try

    config <service> --autoconnect yes

WaifThe3rd, 2016-10-19:
Hello,

So I have a potential workaround that seems to work consistently for me. 

Just doing:
config wifi_alksdj_20934203_managed_psk --autoconnect yes

Doesn't seem to work. still not autoconnecting or pulling into the list of services.

So, I have enabled:
lakka:~ # systemctl enable retroarch-autostart.service

and then
lakka:~ # cd /storage/.config/
nano autostart.sh

#!/bin/bash
sleep 30s <- changing the sleep values lower for me the wifi doesnt connect. Seems like it needs some time to find the wifi adapter / network correctly. 
connmanctl scan wifi
sleep 1m <- changing the sleep values lower for me the wifi doesnt connect. Seems like it needs some time to find the wifi adapter / network correctly. 
connmanctl connect wifi_alksdj_20934203_managed_psk
exit

Checked the startup script for errors:
lakka:~ # systemctl status retroarch-autostart.service

No errors happen, and connection connects, and ip is pingable. 

This seems to work for me for now. Just to note, doing the autostart.sh script by itself didn't work either.

Only changing the config wifi_lajkdsflaks_34234234_managed_psk --auotconnect yes and adding the startup script was I able to make any progress at all.

Seems like its hit or miss with this adapter when it tries to autoconnect, or find it, not sure what though.

WaifThe3rd, 2016-10-19:
Well trying this after i removed the ethernet cable doesn't seem to work now, only worked when I had the ethernet cable inserted for some reason. back to square 1.

WaifThe3rd, 2016-10-19:
Well i found my BIOs had wireless LAN disabled, so I enabled that, removed the autostart stuff, and eventually it connects to the wife, but takes a loooong time. anyways I guess this works, just a lil slow for some reason, maybe its my adapter being flaky.