libretro/Lakka #218: Sound+VYSYNC? Errors

eXcitedlyapathetic, 2015-07-02:
First off, thank you for an excellent job, and all the hard work put into these releases.  Also I apologize if I shouldn't be posting here with this sort of issue, as it may be my lack of experience.

I thought I could put something together for my nephew to play some of the older retro games, but the hardware is pretty barebones (HP Mini 110-3030nr - specs below)
NES/GBA/Genesis all seem to play without a problem, but SNES has alot of stuttering to it(Audio and Video), it's almost worse than the PS1 (which I expected would not play at all).
I'm guessing this is hardware limitations, but I wanted to confirm before I scrap the project.

Thanks again.  - Specs below

Screen Size 10.1 inches
Screen Resolution 1024 x 600
Max Screen Resolution 1024 x 600 pixels
Processor1.66 Hz Intel Atom N450
RAM	1 GB SDRAM
Memory Speed	667 MHz
Hard Drive	160 GB SATA
Graphics Coprocessor	Intel GMA
Chipset Brand	Intel
Graphics Media Accelerator 3150
Graphics Card Ram Size	256 MB

Kivutar, 2015-07-02:
Hello, in the PC version, we ship two snes cores. One is optimized for accuracy (bsnes) and the other is optimized for speed (snes9x). Please make sure to use the second one. You should have two snes icons in your list.

eXcitedlyapathetic, 2015-07-02:
Beautiful.  Thank you very much.

closed 2015-07-02 by eXcitedlyapathetic