libretro/Lakka #220: FFmpeg struggling in Lakka for Raspberry Pi

xmat123x, 2015-07-12:
Using a Raspberry Pi 1 Model B for testing, playing any video file from USB will cause it to drop to 4 or 5 frames per second. Using the same SD, I flashed back OpenELEC Kodi and can see that the hardware has no trouble handling media playback. 

I understand that playing movies isn't Lakka's primary purpose but I'd be great having it be an all-in-one distro for dirt cheap development boards.

twinaphex, 2015-07-12:
That's probably on account of the ffmpeg libretro core falling back to software rendering.

Figuring out a way to make it do hardware accelerated decoding through some API like VDPAU and all of the derivatives would be ideal, but we need some serious research and development going on to see how feasible it is to achieve that through the libretro API.

Kivutar, 2015-07-13:
I guess we have to read the code of omxplayer, the renderer plugin of kodi for rpi. https://github.com/huceke/omxplayer