libretro/Lakka #221: Package e2fsprogs can't be built

Fighter19, 2015-07-16:
Building e2fsprogs fails at current fresh repo

DIR=/mnt/Lakka/build.Lakka-Generic.x86_64-devel/e2fsprogs-1.42.12/lib/ss _SS_DIR_OVERRIDE=. ./mk_cmds /mnt/Lakka/build.Lakka-Generic.x86_64-devel/e2fsprogs-1.42.12/lib/ss/std_rqs.ct
chmod: ss_err.h: new permissions are r-xr-xrwx, not r-xr-xr-x
chmod: ss_err.c: new permissions are r-xr-xrwx, not r-xr-xr-x
Makefile:414: recipe for target 'ss_err.c' failed
make[3]: *** [ss_err.c] Error 1
make[3]: *** Waiting for unfinished jobs....
chmod: ss_err.c: new permissions are r-xr-xrwx, not r-xr-xr-x
Makefile:414: recipe for target 'ss_err.h' failed
make[3]: *** [ss_err.h] Error 1
chmod: std_rqs.c: new permissions are r-xr-xrwx, not r-xr-xr-x

Kivutar, 2015-10-30:
Hi, I built again from scratch on a debian one month ago, without this issue.

Kivutar, 2015-11-30:
I built from scratch again today on a new debian. No problem with e2fsprogs either. I close this fror now. Feel free to reopen it if it still happens.

closed 2015-11-30 by Kivutar