libretro/Lakka #224: Can't take screeenshots in Raspberry Pi 2

aldatsa, 2015-08-04:
I made a clean install in my Raspberry Pi 2 few days ago using this file:
http://sources.lakka.tv/nightly/RPi2.arm/Lakka-RPi2.arm-devel-20150529173709-r21027-g43f71a8.img.gz

Today I tried to take a screenshot of a game but I couldn't. The screen became black, then the main menu appeared back but I lost my game (I couldn't resume). I checked the screenshots folder but it was empty.

I tried taking screenshots of various games in different emulators (FBA, NES, SNES, GameBoy, Master System...) but I've got the same problem in all of them.

Has anybody else run into the same problem?

Let me know if you need more information.

Kivutar, 2015-08-04:
The version of RetroArch shiped at 2015-05-29 had the screenshot functionnality bugged. It have been fixed since that time. I think.
You can try one of our recent nightly builds to check if screenshot are fixed. They should :)

aldatsa, 2015-08-04:
Thanks for your quick reply!

I downloaded the latest nightly build to test it:
http://sources.lakka.tv/nightly/RPi2.arm/Lakka-RPi2.arm-devel-20150728020009-r21120-gb6541ab.img.gz

Now I can take screenshots :)

closed 2015-08-04 by aldatsa