libretro/Lakka #225: No sound - RPI2 connected to Sony KLV-40V300A

BeresfordHare, 2015-08-11:
I imaged my Pi with **Lakka-RPi2.arm-devel-20150529173709-r21027-g43f71a8.img**

It starts alright, but there is no sound. Sound does work with Recalbox and openELEC. Volume was set to 0 in settings and no audio devices were listed.

Kivutar, 2015-08-11:
Hello, try to check the hdmi_mode in config.txt

BeresfordHare, 2015-08-12:
I uncommented the `hdmi_drive=2` in the config.txt but it made no difference. Is that the setting you were referring to?

Kivutar, 2015-08-12:
Yes, I was referring to that setting.

So, if it didn't work, please try to override the full config.txt of Lakka with one of OpenELEC or Recalbox that was working. I could work.

BeresfordHare, 2015-08-14:
I tried the config.txt from OpenELEC and it didn't make a difference. Plus, it looked almost identical to the one in Lakka, except the Lakka one had the `hdmi_drive` option.

Kivutar, 2015-08-14:
There are not so much differences between OpenELEC and Lakka, especially on RPi2. That's why I feel hard to understand why you get no sound.
I still remember that in the early stages, we had to enable the alsa module in the kernel for the audio card of the rpi. OpenELEC was not using it. But now I think that they do.
Try to override the KERNEL of Lakka by the one of OpenELEC? As a diagnostic.

BeresfordHare, 2015-08-14:
I can't figure out how to replace the kernel. Swapping over the kernel.img file breaks everything.
I noticed with openELEC, audio seems to initialise, and a information bubble pops show
>Pulse-Eight CEC adapter
>Connected: TV

Kivutar, 2015-11-20:
I don't know where this bug comes from. Yesterday, I tried Lakka on RPi2 on a TV and the audio was not working, then I uncommented the hdmi_drive line, and audio started working...

You already tried to adjust the audio_volume I guess.

I don't know what to suggest next...

Kivutar, 2016-01-09:
Can you try again with a clean install of the latest nightly build please? And reopen this issue if you're still affected by this bug after uncommenting hdmi_drive=2.

closed 2016-01-09 by Kivutar