libretro/Lakka #226: Composite video on banana pi

diogoviannaaraujo, 2015-08-13:
Im using lakka nicely on hdmi monitor with my banana pi, but when i try to boot on composite is all blank,
Any suggestions?

Kivutar, 2015-08-13:
Hello, I know that some people got the composite working using an alternate
script.bin (however, you will lose hdmi)

alexandrevicenzi, 2015-09-19:
As @Kivutar said, yes you can use composite video, but you will only use one at time.

If you want to switch from HDMI to composite you will need to edit script.bin, you can do this using [Sunix Tool](http://sunxi.org/Sunxi-tools).

The codes of HDMI and Composite are available [here](http://elinux.org/RPiconfig#Video).

Convert your bin to fex and edit with any text editor:

`./bin2fex script.bin>script.fex`

Generate a new bin:

`./fex2bin script.fex>script.bin`

Kivutar, 2016-01-16:
Anybody remember where we can find a precompiled script.bin with composite support? So I could add it in our wiki and close this.

alexandrevicenzi, 2016-01-16:
@Kivutar I've made by hand once, I could try to do again and paste `script.bin` here. I'll take a look.

alexandrevicenzi, 2016-01-20:
@Kivutar could you see if it works?

This is NTSC 1024x768.

[composite.zip](https://github.com/libretro/Lakka/files/96952/composite.zip)

Kivutar, 2016-05-15:
I don't own a bananapi myself to test it. Can anybody try it please?

Sour-Graping, 2016-06-10:
A bit late for a post but

Didn't work for me as is.

I used that as a base and replaced some values I took from here.

http://webcache.googleusercontent.com/search?q=cache:h7nYXRyCHrIJ:forum.lemaker.org/thread-657-1-1.html+&cd=1&hl=en&ct=clnk&gl=ph

Finally worked but the edges are cut like in overscan.
Tested on Banana Pi M1 using debian wheezy.
It won't accept my .zip file so I'm uploading here instead.
https://www.mediafire.com/?qb1fy6g28m7q6w9

xPathin, 2016-06-13:
Thanks Sour-Graping using your configuration i managed to make one myself for PAL. Any luck with the "overscan"-like issue? I'm facing the same...

Sour-Graping, 2016-06-14:
I haven't found a fix for the overscan issue(tested on LED TV) on mine yet.
I'm not sure which value to tweak for that either.

The overscan doesn't bother me as much with what I need to do(Check why it won't boot, fsck taking too long or why I can't SSH to it) so I probably won't find a fix for it anytime soon.

xPathin, 2016-06-14:
Thanks for the heads up. I solved the issue by buying a Raspberry Pi 3

Kivutar, 2016-06-14:
Brutal but efficient fix.. xD

nwrkbiz, 2017-06-14:
Read here for a composite video overscan fix: https://forum.armbian.com/index.php?/topic/3837-psa-orange-pi-zero-expansion-board-tv-out-not-working-solution/&do=findComment&comment=32699

You may have to adjust the base address for the TV Encoder register.