libretro/Lakka #227: Make compatible with Virtualbox

fmarot, 2015-08-20:
I tried to install Lakka into Virtualbox but the installation hangs (I used the hack enabling VBox to boot from a USB key by creating a vmdk file pointing to the key).
As I understand it, it's a simple matter of including correct graphic drivers (and network ?) in the distribution.
Is it possible ? It would be really great to be able to run Lakka directly from my  desktop without having to dedicate an entire hard drive or having a specific computer dedicated...

Kivutar, 2015-08-20:
Just asking, why not using RetroArch directly on your native OS?

fmarot, 2015-08-20:
Good question ;)
First, last time I tried Retroarch directly I had too much to configure, it did not seem to work right out of the box. From what I understand, Lakka works more out of the box.
Second is that I like the Lakka GUI :)

Kivutar, 2015-08-20:
I understand. The thing is that VirtualBox can be OK to get an idea of what the final result is, but it will ruin your performance. It adds overhead. And the only graphical drivers I found working were drivers for Xorg, and we strive to become Xorg-free again.

The interface of Lakka is part of RetroArch now. You can get the exact same result with just using RetroArch. So I guess we'd better do some packaging effort to make RetroArch more out-of-the-box.

fmarot, 2015-08-21:
As an alternative, would it be possible to make the installer use an existing partition instead of a full disk (if I'm correct right now you have to dedicate a full hard drive to Lakka) ?
The goal would be still to use it without having to dedicate too much hardware.

(please tell me if you want me to create another issue for this new feature request)

Kivutar, 2015-08-21:
It would be less work for everybody if we focused on packaging RetroArch preconfigured. The userbase of RetroArch is larger than the userbase of Lakka, more users would benefit of this.

I suggest you to setup the latest RetroArch on your OS, and list all the features not working out of the box compared to Lakka, and we will fix them. Please declare them on RetroArch wiki.

Modifying the installer of Lakka would cause problems when merging the remote OpenELEC code. We need to avoid becoming a complete fork (because the dev team of Lakka is composed of just me).

fmarot, 2015-08-21:
That totally makes sense (I'm a coder myself). I'm trying retroarch right now and it's not that hard ! ;)

closed 2015-10-30 by Kivutar