libretro/Lakka #228: Lakka has some problems when plugging two USB controllers

marcosps, 2015-09-18:
Just one of them is active... I also tried to remove and then replug them, and with this tricky way I could make one of then working, and the other was partly working: the r-pad was working nicely, bu then the d-pad wasn't.

There is a way to make both of then to work? Everybody wants to play versus :)

Thanks!

alexandrevicenzi, 2015-09-18:
What controller are you using?

Could you send the joypad conf (http://www.lakka.tv/doc/Contributing-your-joypad-config/)?

marcosps, 2015-09-18:
@alexandrevicenzi thanks for the links :)

I'll check right now and then report here if it worked! Thanks again!

marcosps, 2015-09-18:
Some observations: if I plug first the "DragonRise Inc.   Generic   USB  Joystick   (0079:0006)", enter in a game and after plug my "0810:0001 Personal Communication Systems, Inc. Dual PSX Adaptor", it works.

But, if I start lakka, plug both controllers and after start a game, the DragonRise is not recognized (but I can see a "Port2: Dragon... configured").

is there a way to fix this misbehavior? Please let me now if you need more info.

Thanks!

Kivutar, 2015-09-18:
I also have a DraginRise controller, and it takes like 1 minute to be recognized (whatever the order of plugging)

marcosps, 2015-09-18:
hum..

Jean, I`ll try to plug it, and wait one minute to verify if this works!

Thanks!

Kivutar, 2016-01-09:
I close this, as I can play with 2 or 3 controllers without problems. This bug may be a controller specific issue, and I don't think we can fix it if it's really the case.

You can reopen this issue if you have more informations.

closed 2016-01-09 by Kivutar