libretro/Lakka #229: [Enhancement] GUI for WiFi configuration

Gronis, 2015-09-20:
Lakka features WiFi configuration via CLI only, which is not optimal for non debug devices. It would be preferable to configure WiFi with a GUI instead. Also, an on screen keyboard would be preferable since most devices that use lakka probably won't have a keyboard easily assessable. I know this would not be that easy to implement, but the current configuration is not user friendly, and terrible to use on end devices.

Kivutar, 2015-09-20:
I agree

Gronis, 2015-09-22:
Is there anyway I can help? Is there a guide how to setup an environment and where to access code documentation? Perhaps this is more relevant to the retroArch project rather than lakka project?

Kivutar, 2015-09-22:
This kind of development have to be done in RetroArch, but will be useful just for Lakka.

We use Connman as connexion manager. So we have to expand the GUI of RetroArch and allow it to write Connman config files.

The documentation to read here is the one of Connman. RetroArch have nothing special, just configure && make

Kivutar, 2016-10-02:
Ok this is done in the development builds.

closed 2016-10-02 by Kivutar

manast, 2017-01-05:
I have a strange problem where the screen keyboard pops-up for entering the Wifi passphrase, but after that lakka freezes, or at least it stops taking input from the keyboard, I have tries with mouse, joysticks, etc. but as soon as the screen keyboard appears I need to reboot in order to regain control. The same issue appears in other options such as audio. Seems like the screen keyboard just do not work on my device. Any ideas?

manast, 2017-01-05:
ok, I managed to enter the passphrase using a ps3 controller... still strange that it did not work with a standard keyboard

HerbFargus, 2017-03-11:
I can also attest to the keyboard not working for some reason on my odroid. c1+ but works fine with my Logitech controller