libretro/Lakka #230: Menu on Cubox-i is not looking as expected

oblitum, 2015-10-16:
I've tried an early Lakka build from May and the menu resembled the screenshots I've seen around, but I just updated the system to the latest release and the menu is looking like the following, is this right?

[image]

[image]

oblitum, 2015-10-16:
I just get vertical tree access, nothing like the original horizontal menu with the consoles.

Kivutar, 2015-10-16:
Ah, you are using the development version. Here you have to add content using the add content menu entry. It will generate the lists you need.

oblitum, 2015-10-16:
@Kivutar Thanks! I'll try it when I get back home and close it if OK =)

An offtopic question, this morning I've tried to shutdown lakka selecting shutdown with the xbox wireless joystick but lakka just freeze on shutdown, I've to unplug it from the wall.

oblitum, 2015-10-16:
I mean, on clicking shutdown, nothing happens and I can't do anything in the UI.

oblitum, 2015-10-16:
If this is a non-reported issue I may open another one to register it.

Kivutar, 2015-10-16:
For me, it takes 20 seconds sometimes, but then it shut down

oblitum, 2015-10-16:
In my case it didn't, I've wait way more than 20 seconds, will try again when back home.

Kivutar, 2015-10-16:
In cuboxi, the system fails to close the display. Check if the LEDs of the cuboxi are telling about the shutdown?

oblitum, 2015-10-17:
Hi @Kivutar, the options I have at "Add Content" don't help, "Download Content" turn the UI irresponsive just like when I get a freeze shutting down, the clock keeps changing but I'm unable to interact with the UI, I may reboot from ssh though. The other options, "Scan Directory" and "Scan File", are empty.

oblitum, 2015-10-17:
I'd be glad to know how to put this "Content" manually.

Kivutar, 2015-10-17:
First, transfer your roms to the rom directory.
Then, use the Add Content entry to scan this directory.
Try with super nintendo games.

oblitum, 2015-10-17:
@Kivutar This worked t show up the consoles! I've copied 3 Mario games for 3 nintendo consoles and the 3 joystick icons shown up. There's still a problem though, when I select a game from this menu, there's the drop down option to Run the game but I can't run it, this option says "No cores available".

oblitum, 2015-10-17:
@Kivutar Nice, I've run all the updates in the online updater section, "quit RetroArch", and now it's working.

closed 2015-10-17 by oblitum