libretro/Lakka #231: rescanning content create duplicates

oblitum, 2015-10-19:
One can have a big issue having a big list of games in `roms` scanned and then ask a rescan of the directory, duplicates of all roms in there will be created in the main menu, one then needs to delete the playlists file with the duplicates that populate the menu and ask for a rescan from zero. Currently when I have new games, I move the old roms directory to a new one, copy the new roms to a empty roms directory, scan it so that only new games are scanned, then move back the old roms.

sergiobenrocha2, 2015-11-28:
https://github.com/libretro/RetroArch/issues/1908

fabioxgn, 2016-01-12:
I have the same issue, how do I reset the games list so I can scan again?

oblitum, 2016-01-12:
@fabioxgn if you have access to the device file structure, you can just remove the playlists directory.

oblitum, 2016-01-12:
btw, I've generated my playlists "by hand" scripts, because of #232.

Kivutar, 2016-01-29:
Fixed by https://github.com/libretro/RetroArch/commit/2af07080479cbaf7a8cbb72edb0573c3a2e8172d

closed 2016-01-29 by Kivutar