libretro/Lakka #232: scanning process to generate playlists fails for some roms without apparent reason

oblitum, 2015-10-19:
Example, [this rom set](https://thepiratebay.mn/torrent/4664803/NINTENDO_64_COMPLETE_%28U%29_%5B_%5D_ROMSET) has a `Mario Kart 64 (U) [!].zip` rom, I unzip it and copy `Mario Kart 64 (U) [!].z64` to the roms directory, I then go to "Add Content" and scan the roms directory, the rom doesn't get listed in the menus afterwards but if I go to "Load Core" and load mupen64plus and then chose to load a rom from a file, the rom is loadable, so the sole thing not working is the scan process.

(by the way, mupen64plus is unplayable in the 4-core ARM processor of cubox-i, if anyone got any tip, I suspect the emulator is not good enough compared to PJ64)

Kivutar, 2015-10-19:
Certainly due to an incomplete database. Maybe the database doesn't contain the checksums for z64 roms, but n64 roms would work?

oblitum, 2015-10-19:
I'm gonna check this.

oblitum, 2015-10-20:
The database seems unreliable as a source, at last for me, just tried to scan an PSX iso without success, I was then able to load it from file after loading the PSX core. I'd be very glad if there was an alternative method for building a playlist menu.

oblitum, 2015-10-20:
Like, just associating file extension to respective consoles and listing their names per console.

oblitum, 2015-10-20:
I guess I'm gonna check the playlist structure and build them "by hand".

Kivutar, 2015-10-20:
Associating file extensions to console is what we had before, and it caused problems due to:

 * arcade roms having to be .zip
 * .bin being a very generic extension among emulators
 * various CD emulators using .cue, .img, .bin, .iso

oblitum, 2015-10-20:
It seems a directory structure for the roms directory would offer a simple solution maybe, just having subdirectories per emulator, it's better than not being able to list the games because of the CRC/database method.

oblitum, 2015-10-20:
A current entry in the playlists follow this pattern:

```
/storage/roms/Super Mario 64 (USA).n64
Super Mario 64 (USA)
DETECT
DETECT
42C43204|crc
Nintendo - Nintendo 64.lpl
```

I'm thinking of generating lists programmatically with a custom method, but I'm not sure whether the generated playlists will provide good entries in the menu.

Can you please confirm whether this is feasible, for example, I currently have no idea what value to put for `crc` for the games I have, it's the sole thing that's unclear, and also I'm not sure this crc means that a playlist entry must have a crc that exists in the /tmp/database files, which I'm thinking of ignoring.

Kivutar, 2015-10-20:
It is feasible, I'm trying this at the moment:

```
#!/usr/bin/bash

for fullpath in $1; do
	echo $fullpath
	filename=$(basename "$fullpath")
	echo "${filename%.*}"
	echo $2
	echo $3
	echo "DETECT"
	echo $4
done
```

And I call it like this:

    ./scan.sh "/home/kivutar/Jeux/roms/snes/*.smc" /usr/lib/libretro/bsnes_accuracy_libretro.so "bsnes Accuracy" "Nintendo - Super Nintendo Entertainment System.lpl" > playlists/Nintendo\ -\ Super\ Nintendo\ Entertainment\ System.lpl

fr500, 2015-10-20:
"It seems a directory structure for the roms directory would offer a simple solution maybe, just having subdirectories per emulator, it's better than not being able to list the games because of the CRC/database method."

This would imply hardcoding what emulators are for what platform in RetroArch or something like that.
RetroArch MUST stay core independent, no hacks to bind ANY kind of functionality to a particular core.

oblitum, 2015-10-20:
@fr500 not necessarily "per emulator", it could be "per plataform", for example, I'm going to build playlists myself, and it will be per platform, the association of the emulator is the final hardcoded part. A directory structure could be build per platform, and for each platform a given core could be associated.

oblitum, 2015-10-20:
@Kivutar Thanks for the info!

oblitum, 2015-10-20:
@Kivutar I'm only in doubt about the "bsnes Accuracy" argument, I'll grep it later but if you could explain it to me so that I replicate for other platforms, I'd be glad.

oblitum, 2015-10-20:
@Kivutar ah, it's the emulator name, now I got it, nevermind, thanks.

oblitum, 2015-10-21:
Hi @Kivutar, `/usr/lib/libretro/bsnes_accuracy_libretro.so`, as you provided in your command, is not a file that came with the system, any issue with that? There're few .so's in `/usr/lib/libretro`, mostly are info files.

oblitum, 2015-10-21:
a nevermind again, I guess I should use the one available there, which is already working, snes9x_next_libretro.so

oblitum, 2015-10-21:
At last, after some scripts, I managed to build a working playlist... =)

Kivutar, 2015-10-30:
You can get the rom sets called "no-intro", those one are very clean will be scanned 100%. Our database comes from the no-intro project.

Kivutar, 2015-11-20:
I tried all the no-intro sets and they scan well. I also updated our documentation to invite users to use these dumps. I close this.

closed 2015-11-20 by Kivutar

oblitum, 2016-09-22:
I've downloaded a bunch of no-intro sets from <http://www.epforums.org/showthread.php?56293>. I have noticed that the Atari sets from there (7800 and Lynx at last) are not being recognized at all, only a few. I dunno how no-intro is being used at a reference for scanning as a whole, but it's being a bit hard and confuse to get a scanning that works.

cudencuden, 2016-09-23:
that thread has sets with mixtures of dates..some from couple years back.....start with a 2016 into set...it's freely available...use Google

Kivutar, 2016-09-23:
Take no-intro roms from 2016 and done.

oblitum, 2016-09-23:
OK. I hope I'm downloading from the right place now...