libretro/Lakka #233: Provide easy mode?

oblitum, 2015-10-21:
Hi, after some struggle I managed to configure a lot of games for several emulator for my little cubox-i, I'd like to share this device with my family, including children and nani, so after setting up the device I'd be very glad with an option to hide the configuration menu, so that only consoles and their games gets listed.

If there's any simple hack (through ssh or mounting the memory card) without recompilation that I may try for that, I'd like to know.

Kivutar, 2015-10-21:
Hello, I understand your request.
I think I can code something like that, like a boolean setting accessible only from retroarch.cfg
@twinaphex what do you think about it? Is it acceptable in RA?

oblitum, 2015-10-21:
Thanks, it would be awesome =)

ochief, 2015-11-09:
I reckon a very simple way of achieving this would be to add a flag for 'kiosk' mode or something similar to retroarch.cfg and then you would only need to make the following changes to the menus for it  to provide a tamper proof experience based on that flag.

Settings Menu
-------------------
Hide All

This way it would immediately boot into your playlists.  Even though this would stop you from accessing the load content options I think this 'easy mode/kiosk' only makes sense for a fully configured system.  If you needed to update cores etc. you could just go back into the normal mode.

Quick Menu
-------------------
Add Quit Retroarch
Add Shutdown
Hide Core Options
Hide Core Cheat Options
Hide Shader Options

Kivutar, 2015-12-15:
I like the idee of the kiosk mode.

tirogalo, 2016-02-12:
This is really interesting... is there any progress on this?

Kivutar, 2016-02-12:
Not yet

Kivutar, 2016-08-07:
https://github.com/libretro/RetroArch/pull/3341

Kivutar, 2016-09-24:
Ok, in our development builds, we now have Show Advanced Settings set to false. And we also have a way to hide the Settings tab. The combination of the two should make your system difficult to break.

closed 2016-09-24 by Kivutar

oblitum, 2016-09-29:
@Kivutar any idea about when a new cubox-i release will be available?

oblitum, 2016-09-29:
that will include that I mean.

Kivutar, 2016-10-01:
Yes