libretro/Lakka #234: core updater for ARM not working

cudencuden, 2015-10-22:
using xu3 October 19 2015 build. selecting core updater flashes the screen n nothing

cudencuden, 2015-10-22:
*nothing happens...

cudencuden, 2015-10-22:
in retroarch.cfg ..i see the two lines below...should core_updater_buildbot_url be blank?

core_updater_buildbot_url = ""
core_updater_buildbot_assets_url = "http://buildbot.libretro.com/assets/"

Kivutar, 2015-10-22:
You're right.

cudencuden, 2015-10-22:
I changed to "core_updater_buildbot_url = "http://buildbot.libretro.com/nightly/linux/armhf/latest" about rebooted...but it does the same thing..screen flicker once but nothing happens when i select core updater

cudencuden, 2015-10-22:
i am opening the \\LAKKA\Configfiles\retroarch\retroarch.cfg with notepad (windows machine) over the network and make the change. then when i restart lakka...retroarch.cfg did not save.....so how are you suppose to make edits to retroarch.cfg? via ssh?

cudencuden, 2015-10-23:
I'm pretty sure one of the other online update options mess upcore update....because core updater worked when I used it right after fresh install on x64...then I do other online updates n now core updater does not work..same issue as above

Kivutar, 2015-10-23:
When you edit the configuration file, you have to first stop the retroarch service. If not, retroarch will keep the old configuration in memory, and save it to the file when quitting. Our documentation explain the process to edit that file.

One thing you can do, to start on a clean basis, is stopping retroarch service, backup retroarch.cfg and remove it, start retroarch with that blank config.

From that, if it doesn't work, stop retroarch service, add the right url in the config file, start retroarch again. And see if it works.

cudencuden, 2015-10-23:
1) ssh into lakka
2) typed: systemctl stop retroarch
3) deleted retroarch.cfg
4) typed: systemctl start retroarch
5) verified blank retroarch.cfg is created
6) systemctl stop retroarch
7) open retroarch.cfg via network share
8) added core_updater_buildbot_url = "http://buildbot.libretro.com/nightly/linux/armhf/latest"
9) systemctl start retroarch 

works now. thanks

Kivutar, 2015-10-23:
Ok, I need to understand why this value is not here by default then. Thank you for that test.

cudencuden, 2015-11-25:
hi kivutar...I noticed that everytime I use online updater > autoconfig profile update....I have to quit retroarch for core updater to use the correct url

Kivutar, 2015-11-25:
Well spotted. Let me declare this on the retroarch tracker, where it has more chances to be fixed.

Kivutar, 2015-11-30:
The last cores I tried from the buildbot were wrong. pcsx_rearmed was launching, but segfaulted after sometimes.

cudencuden, 2015-11-30:
so the source files at http://buildbot.libretro.com/nightly/linux/armhf/latest is wrong?

fr500, 2015-11-30:
those are binaries, not sources

Kivutar, 2015-12-01:
Yes those are binaries. On beetle-psx, the shadows on 3d models were wrong, and it segfaulted on the title screen of Crash Bandicoot. Maybe an incompatibilité in the optimizations we enabled at compile time?

Kivutar, 2016-01-09:
Fixed by 4dadcab37a2dd7703e18c81b5add951197a9a7eb

closed 2016-01-09 by Kivutar

xeros, 2016-02-24:
Thx for patching it, but now the buildbot should be fixed: https://github.com/libretro/libretro-buildbot/issues/5