libretro/Lakka #235: BleagleBone Support

alexandrevicenzi, 2015-10-22:
Hi,

Is it possible to run Lakka on BeagleBone (Green or Black)? There's no support yet.

This is just a question, I'm not asking for a new version, but it would be great. :smile:

Kivutar, 2015-10-22:
Hello Alexandre,

None of our current builds supports the processors of the BeagleBones.

However, if somebody sends us a free sample, I would be happy to port Lakka to this platform.

alexandrevicenzi, 2015-10-22:
I have a Green one, but I don't have a lot of free time. What would be necessary to port Lakka?

Kivutar, 2015-10-22:
You need to find these informations:

 * what gcc flags are used to cross compile for this specific processor
 * what bootloader they use, if it's uboot, what branch of what repo they use
 * what kernel they use, the repo, the branch, the kernel defconfig
 * an fbdev+gles capable gpu userspace driver

alexandrevicenzi, 2015-10-22:
@Kivutar I'll try to take a look. Thanks.

Kivutar, 2016-03-14:
I close this for now. If somebody want to context BeagleBone to get a free unit, I will do that port.

closed 2016-03-14 by Kivutar

kennonkwok, 2016-04-09:
@Kivutar I can send you a BBB that I have laying around.  Is there someone in the US to receive it? PM me.

Kivutar, 2016-04-09:
Hi @kennonkwok 

Thanks a lot.

I'm currently in Vietnam. I will be in Europe this summer in theory.

jacobblock, 2016-11-28:
Hi did this ever happen?

kennonkwok, 2016-11-28:
No, I don't have a good way to get it to either VN or EU without paying almost the cost of the BBB in shipping :-)

jacobblock, 2016-11-28:
Thanks, I am interested in this too so I will try and find some time this December to contribute.

jacobblock, 2016-11-30:
Fairly new to all of this, but here is a very good page on the cross compiling build environment for BBB:

https://eewiki.net/display/linuxonarm/BeagleBone+Black

Compiler: gcc-linaro-5.3.1-2016.05-x86_64_arm-linux-gnueabihf
U-Boot: https://github.com/u-boot/u-boot branch v2016.11-rc3
Kernel: am33x-4.X (latest and greatest if wanted)
defconfig: See patch here https://eewiki.net/display/linuxonarm/BeagleBone+Black#BeagleBoneBlack-Bootloader:U-Boot
fbdev driver: xserver-xorg-video-fbdev
GLES driver: SGX modules with TI's GFX 5.01.01.02. There is also some discussions here from last year: https://groups.google.com/forum/#!topic/beagleboard/tsRfbDVwjSM

Kivutar, 2016-11-30:
Prefer the fbdev or drm/kms driver over the xorg one