libretro/Lakka #236: Wrong colors with Mame emulation

guifes, 2015-11-03:
Almost every game I try to emulate in MAME plays with wrong colors.

Kivutar, 2015-11-03:
Hello, what hardware are you using? What version of Lakka? and can we get a screenshot? Thank you!

guifes, 2015-11-05:
Sorry it took me long. The pc @ my workplace. I'm probably running the last version of Lakka, since I downloaded it a few days ago. About the hardware I'll have to check (I'll try and post it here asap with the screenshot as well). But the thing is some games do work perfectly. Almost every game that needs neogeo.zip is not running at all, and I have put neogeo.zip in the rooms folder. (I've put the roms in a directory MAME inside the roms folder would that be a problem?). The games running with wrong colors are TMNT and Simpsons.

Kivutar, 2015-11-05:
You're using the only mame packaged in Lakka? (the old mame2003) or one downloaded from the Core Updater?

guifes, 2015-11-05:
I did not download aything besides lakka, and haven't updated anything as well.

Kivutar, 2015-11-05:
Can you try the core Dinothawr or Nestopia? They use a 32bit color mode. Maybe you graphical card doesn't support this mode. The mame2003 switches to 32bit depending on the game.

guifes, 2015-11-05:
How exactly do I do that?

Kivutar, 2015-11-06:
Just load a NES game.

Kivutar, 2016-03-14:
@guifes any update on this? (Please re-open the issue when you have tried launching a NES game)

closed 2016-03-14 by Kivutar