libretro/Lakka #237: ATI HDMI Audio not working in r21189

matcarfer, 2015-11-05:
HDMI audio is not working. Even disabling integrated audio. When using aplay -L it shows null or HDMI, still no audio. Also tried manually editing retroarch.cfg and despite all seems ok, its not working. I used help from here: http://libretro.com/forums/showthread.php?t=3531 , here: http://www.lakka.tv/doc/Configuring-Lakka/ , here: http://www.lakka.tv/doc/Audio-settings/ and here: https://github.com/libretro/Lakka/issues/108

HDMI audio is super important, maybe 1st priority? When I meant integrated audio, I meant that PCs with an integrated video card with HDMI port normally have 2 sound devices: HDMI audio device and integrated audio device (analog+optical/coaxial) which normally is a Realtek or VIA device. When disabling integrated audio device, HDMI is the only available, so it should work out of the box, but it doesnt work with or without.

Kivutar, 2015-12-24:
Have you tried this? https://bbs.archlinux.org/viewtopic.php?id=132938

Kivutar, 2016-01-09:
At least one user on IRC solved it using radeon.audio=1 in the cmdline. Feel free to re-open this issue if it doesn't work for you.

closed 2016-01-09 by Kivutar

aboutafter, 2016-01-20:
radeon.audio=1 doesn't fix it for me. I changed audio_device="hdmi:CARD=HDMI,DEV=0" and I don't have sound.