libretro/Lakka #239: Console mode not working with raspberry pi 2

Gronis, 2015-11-06:
Im' trying to enter console mode by adding "tty retroarch=0" to cmdline.txt. It works at first, but by the time when retroarch gui should be displayed during a normal boot, the screen just turns black (I get the terminal a few seconds and can type commands, but then it disappears). I tried to do other things to the cmdline.txt file, like removing quiet option, but the screen always goes black after a while. Also, I'm using a composite output, not hdmi. Can this be a deciding factor?

How do I fix this? I want to be able to enter the terminal without ethernet, because I'm going to desolder the ethernet port and make it portable, so it's quite a big concern.

Kivutar, 2015-11-06:
Did you try Alt+F3?

Gronis, 2015-11-06:
I tried using Ctrl+Alt+F3 (not alt+F3) as shown in the <a href="https://github.com/libretro/Lakka/wiki/Accessing-Lakka-command-line-interface">documentation</a> and it takes me to the terminal. However, after 4 seconds (when the gui would be initialised) the screen just turns black. After Ctrl+Alt+F3, I get to the terminal and i tried to type some commands before 4s has elapsed, and it does work. The problem is that the screen just turns black after the boot has finished somehow. It seams that the terminal is working, but something "takes over" the screen at the end of the boot sequence. Can it be that I use rgui instead of the standard xmb interface?

Kivutar, 2015-11-07:
Ok, I will test that. No it can't be due to rgui.

Gronis, 2015-11-07:
Ok. I will try it out more as well, I post if I find a solution.

Kivutar, 2015-12-01:
I confirm this bug.

Kivutar, 2016-01-09:
Looks like even OpenELEC can't support it http://wiki.openelec.tv/index.php/Boot_Arguments

Gronis, 2016-01-14:
No wonder it's not working for Lakka then. Anyway, you can just enter the pi via ssh to access the terminal so it's not a huge deal. Maybe you should write down that the debug mode is not working for the RPI in the documentation?

Kivutar, 2016-01-14:
Done :)

closed 2016-01-14 by Kivutar

100ideas, 2016-12-18:
According to http://wiki.openelec.tv/index.php/Boot_Arguments, you can pass 'textmode' to get a local terminal. It works for me w/ a logitech usb-dongle-bluetooth keyboard.

Not sure if the normal filesystem gets loaded with this terminal or not.